nxp-imx/mfgtools #280: New release due to a fix for M1 macs

dzhlobo, 2021-08-14:
I'm creating a formula for uuu in homebrew but I got an issue building the latest released version on mac with M1 chip: https://github.com/Homebrew/homebrew-core/pull/83254.

It is already fixed in master on June 29: https://github.com/NXPmicro/mfgtools/issues/274.

Can you please release a new version?

nxpfrankli, 2021-08-16:
try https://github.com/NXPmicro/mfgtools/releases/tag/uuu_1.4.149

dzhlobo, 2021-08-18:
@nxpfrankli, thank you ❤️. I can successfully build homebrew formula locally and uuu seems working. Unfortunately, it seems pre-release versions aren't allowed in homebrew-core:
```
$ brew audit --new uuu
uuu:
  * uuu_1.4.149 is a GitHub pre-release.
Error: 1 problem in 1 formula detected
```

So I think I'll have to wait until it become "latest" version.

nxpfrankli, 2021-08-18:
it should be not difference between pre-release and formal release. 
Pre-release just tag in github. there are not "pre-release" information in git and tarball.

dzhlobo, 2021-08-18:
@nxpfrankli, yes, there's no difference in git and tarball but homebrew has additional checks specifically for GitHub. It parses tarball url, detects that it is from GitHub and pulls meta information about the GitHub release.

It wouldn't be a problem if I upgrade version of an existing package but homebrew is much more strict to new packages. I've also posted a comment in the pull request to homebrew-core, maybe maintainers will comment.