nxp-imx/mfgtools #282: Does uuu support u-boot with plugin embeded?

shawnguo2, 2021-08-15:
I have an i.MX7D based device where u-boot is built with plugin rather than DCD.  Does uuu support booting such u-boot image?  If yes, what would the uuu command/script look like?  Thanks!

nxpfrankli, 2021-08-16:
try 
SDP: write -f uboot -ivt 0
SDP: jump -f uboot -ivt 0
SDP: write -f uboot -ivt 1
SDP: jump -f uboot -ivt 1

nxpfrankli, 2021-08-16:
Sorry, I just check code. Look like not support yet.

nxpfrankli, 2021-08-19:
I just fix implement's bug. Please pull latest uuu. 

Prebuild image is in 

Linux https://ci.appveyor.com/project/nxpfrankli/mfgtools-kvqcg/builds/40447910/job/bm81rgl353ogd4kd/artifacts
windows 
https://ci.appveyor.com/project/nxpfrankli/mfgtools-kvqcg/builds/40447910/job/y3a4lq1uwfm56ub6/artifacts

SDP: write -f uboot -ivt 0
SDP: jump -f uboot -ivt 0
SDP: write -f uboot -ivt 1
SDP: jump -f uboot -ivt 1

shawnguo2, 2021-08-20:
Yes, it's working :+1: Thanks, Frank!

closed 2021-08-20 by shawnguo2