nxp-imx/mfgtools #283: Unable to flash dev kit on Windows with isolated USB interface

EffectiveUA, 2021-09-02:
Hello,

I am observing an issue while flashing my custom PCBA (iMX8M Mini SoM) using UUU tool on Windows.
My PCBA has an isolated USB interface.

Let me briefly describe when the issue occurs:
1) UUU Starts build-in iMX8 loader (HID)
2) Uploads My FW bootloader into iMX8 RAM
3) Restarts with My bootloader -> windows detects new fastboot device in device manager
==> WinUSB driver fails to strat&recognize the device
 
This issue on Windows is not reproducible using dev kit that has the USB interface NOT isolated
The issue is not reproducible on Linux (using my PCBA with isolated USB)

Feel free to revert for any details. Thanks in advance.

nxpfrankli, 2021-09-02:
Look like WCID have not work. 
You use https://zadig.akeo.ie/ to install winusb driver manually.

EffectiveUA, 2021-09-03:
Hi @nxpfrankli !
Thank you for the suggestion!
I should have specified that I'm using Windows 10.

I have tried a ciuple of drivers using the tool you suggest, but none of them worked for me.

Could you suggest a particular driver version?

Thank you in advance!

nxpfrankli, 2021-09-03:
It is not relate version. I look at your screen shot again. winusb driver already loaded and configuration success. 
Look like something block it.

EffectiveUA, 2021-09-03:
That is correct.
I assume that the USB isolation does not allow Windows driver to communicate with device.

What is interesting - this issue is not reproducible with my device on Linux.