nxp-imx/mfgtools #287: Unable to flash SB image using NAND with iMXRT1050

tonyponyexpress, 2021-09-10:
I've tried adding the ROM info and Pid and Vid for the 1050 in order to flash SB images but the script loops on the following command: "Cmd:SDP: boot -f boot_image.sb" Has anyone had any success flashing SB images with the MXRT1050?

lznuaa, 2021-09-10:
Can you try SDPS: boot -f boot_image.sb.   supposed sb file is the same as i.mx28's image

tonyponyexpress, 2021-09-15:
Thanks for the response. After changing it to use the SDPS command, the uuu script doesn't progress on to the next step in flashing. The following is how I'm executing uuu: "sudo uuu -b nand boot_image.sb"

nxpfrankli, 2021-09-15:
write nand is not so easy. 
boot_image.sb need support android fastboot protocol, which communicate with uuu. 

Firstly, does boot_image.sb run at iMXRT1050 with sdps?

tonyponyexpress, 2021-09-15:
Yes, it runs. So, there is no support for flashing sb files at all?

nxpfrankli, 2021-09-15:
I have not worked on iMXRT project. It needs bootloader(boot_image) do some extra work to flash work.