nxp-imx/mfgtools #289: Recovery Process stuck in u-boot on nitrogen8m

licosec, 2021-09-16:
Dear all,

I am currently trying to recover a Boundary Devices Nitrogen8m using uuu. I have followed this tutorial (https://boundarydevices.com/recovering-i-mx-platforms-using-uuu/). As suggested I used uuu_1.3.154 and the u-boot.nitrogen88m provided here (http://linode.boundarydevices.com/u-boot-images/). Additionally I tried with uuu_1.4.139 and a custom built u-boot.nitrogen8m that was bootable on another board.
For all setups I faced the following Problem: The boot process seemed to be stuck somewhere in Uboot. 
The Output from the board was the following:

> U-Boot SPL 2020.10-52989-g5c1c757368 (Sep 10 2021 - 11:02:34 -0700)
> Setting voltages
> config to do 3200 1d training.
> config to do 667 1d training.
> config to do 3200 2d training.
> Normal Boot
> Trying to boot from USB SDP
> SDP: initialize...
> SDP: handle requests...
> Downloading file of size 1132928 to 0x40400000... done
> Jumping to header at 0x40400000
> Header Tag is not an IMX image
> Found header at 0x4040ea00
> 
> 
> U-Boot 2020.10-52989-g5c1c757368 (Sep 10 2021 - 11:02:34 -0700), Build: jenkins-uboot_v2
> 020.10-141
> 
> CPU:   i.MX8MQ rev2.0 at 1000 MHz
> Reset cause: POR
> Model: Boundary Devices i.MX8MQ Nitrogen8M
> Board: nitrogen8m
>        Watchdog enabled
> DRAM:  2 GiB
> MMC:   FSL_SDHC: 0
> Loading Environment from MMC...
> OK
> Display: hdmi:1920x1080M@60 (1920x1080)
> [*]-Video Link 0cdn_api_checkalive: keep-alive counter did not increment for 10us...
> HDMI enable failed, ret -1!
>  (1920 x 1080)
>         [0] display-controller@32e00000, video
>         [1] hdmi@32c00000, display
> In:    serial
> Out:   serial
> Err:   serial
> 
>  BuildInfo:
>   - ATF 63a678c
> 
> Starting fastboot...
> Saving Environment to MMC... Writing to MMC(0)... OK
> Net:   AR8035 at 4
> FEC [PRIME], usb_ether

The output from uuu was:

> $ sudo ./uuu -d u-boot.nitrogen8m_latest 
> uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.3.154-0-g47ddef5
> 
> Success 2    Failure 0                                                                 
>                                                                                         
>                                                                                         
> 1:4      3/ 3 [Done                                  ] SDPV: done   
> 

Has someone experienced a similar issue or has any idea what could have caused it?

Thank you and best regards

licosec, 2021-09-21:
The Ctrl-C did the trick to reach the u-boot console. Thank you very much!

licosec, 2021-09-21:
Unfortunaly it didn't last long. I was able to get to the u-boot console after pressing Ctrl-C and then flash a working u-boot on my board. The problem that it froze at "FEC [PRIME], usb_ether" persisted, even for the flashed U-Boot. Running "fastboot 0" also seemed to freeze and had to be interrupted with Ctrl-C. I settled for the Ctrl-C option to get my shell and boot, however after a few reboots, it is not possible anymore to reach the u-boot console. After pressing Ctrl-C the board remains frozen, in the recovery mode as well as in normal boot mode. Do you have any idea what my have caused that, @nxpfrankli ?

nxpfrankli, 2021-09-21:
It is your uboot behavior. Your uboot look like try download image from network or usb_ether, instead of boot from storage. 
Please check your uboot environment variable.  Which is not related uuu tools.

licosec, 2021-09-22:
Ok, thank you for the hint! How is it possible to access the environment variables without a u-boot shell? I only know of and found the way with 'print', 'setenv' and 'saveenv' from the u-boot console.

nxpfrankli, 2021-09-22:
Some variable saved in emmc special location, or a file at file partition.  And u-boot have default environment. 
It is highly customized.  Efforts of analyze these data is quite high without u-boot console.

licosec, 2021-09-23:
Ok, good to know, thank you.
I'll close this now since it's not really uuu related.

closed 2021-09-23 by licosec