nxp-imx/mfgtools #291: Relation between UUU2 flashing and Windows OS performance #270

rajumallah, 2021-09-29:
Hi All,

https://github.com/NXPmicro/mfgtools/issues/270

Continuing to above issue, 

Can you please provide a list of VID used for Ublox which need to be added in ignored list because we are seeing in registry that there are several other VIDs are registered. it is difficult to search for Ublox id.

Since with every device getting connected to PC registry is growing and it is causing system degradation.
Generally size of registry file is around ~400MB for production pc, but after production of several hundreds of devices connected to USB of pc, its size grown upto 1.5 GB.

Thanks,
Raju Mallah

nxpfrankli, 2021-09-29:
run cmd 
uuu -IgSerNum

nxpfrankli, 2021-12-02:
no feedback over month. if still problem, please reopen

closed 2021-12-02 by nxpfrankli