nxp-imx/mfgtools #292: Bulk(R):LIBUSB_ERROR_TIMEOUT when flashing Android sparse image file via Fastboot

zm33y, 2021-10-01:
We are building a .wic image with Yocto build system, and the resulting file is about 1.6G. When flashing it via Fastboot, using `FB: flash -raw2sparse all image.wic` command it works fine but the process is too slow (about 2.5 min).

Next step we tried is to convert .wic image (a sparse file) into Android sparse file format and flash it without `-raw2sparse` flag.

```
$ img2simg  image.wic  image.android

$ sudo ./uuu uuu.script.android 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.165-0-ge10b026

Success 0    Failure 1  

1:4     12/14 [Bulk(R):LIBUSB_ERROR_TIMEOUT          ] FB: flash all image.android
````

This process is much faster (about 20 sec) but it fails with `Bulk(R):LIBUSB_ERROR_TIMEOUT` error. However the image seems to be flashed fine - after manual reset of the board it boots successfully.

nxpfrankli, 2021-10-01:
img2simg will convert some continue zero to a small trunk.  such write 0 to continue 2M.  but write continue 0 to emmc/sd still take  quite long time. That's reason is why you get timeout.  You should add FB[-t:50000] to increase timeout value. 

After increase timeout, make sure wait for write complete. The total time should be similar,  sparse file only saved data transfer time cross USB.

img2simg don't know filesystem information. If tools can know filesystem information, there will be some NO-CARED data.  then uuu can skip NO-Cared Data for both transfer and write to eMMC.

zm33y, 2021-10-01:
@nxpfrankli thanks for the prompt reply!
Timeout helped:
```
FB[-t 150000]: flash all image.android
```
Now flashing takes 1.5 min vs 2.5 min which is much better. But still longer than it might be.

For example, we also have a .bmap file and flashing an SD card with the same image takes about 15 seconds. I wonder if we could use the same .bmap file and write arbitrary regions on flash via Fastboot? Does Fastboot support writing data at arbitrary offsets?

> If tools can know filesystem information, there will be some NO-CARED data. then uuu can skip NO-Cared Data for both transfer and write to eMMC.

Does this mean we need a separate image for each partition or we can somehow have one .wic image with different file systems with such information per file system?
And how do I get this file system image with such 'NO-CARED data'?

nxpfrankli, 2021-10-01:
It is trick to do arbitrary offset by fastboot. It needs create a no-cared chuck before other data.  it can write to any position by adjust no-cared data size. 

Some android tools can generate it only for ext4 file system. 

I have idea before.  Modern file system support sparse file.  For example,  when yocto generate a 1.6G empty image file, file system know all is empty and only allocate one block for this 1.6G file.  file system allocate physicals block only when application write data into this file.   User can get such information by some API.  So there are possible to write a tools to get these segment data and create an android sparse file. Untouched block can leave NO-Cared. 

Because our release's wic image zero only occupy about 10% data,  I have not written above tools.

nxpfrankli, 2021-10-01:
I just read https://github.com/intel/bmap-tools, which is the same idea. 

maybe there are tools convert bmap into android sparse

zm33y, 2021-10-01:
> So there are possible to write a tools to get these segment data and create an android sparse file. Untouched block can leave NO-Cared.

I think you are describing `img2simg` tool. It converts a sparse file (1.6G) into an android sparse file (56M):
```
# Show 'sparsity' factor:
$ find image.wic -printf '%S\n'
0.0347681
$ stat -c %s image.wic
1691385856     <--  size in bytes
# 0.0347681 * 1691385856 = 56M  <-- actual 'payload' size

# create Android sparse file
$ img2simg  image.wic  image.android
$ stat -c %s image.android 
58152956   <-- 56M
```

So the sparse image is correct. However the programming still takes too long (1 min) even though the amount of data that needs to be written is not that big.

nxpfrankli, 2021-10-01:
I read img2simg source some years ago.  img2simg just check continue zero.  for example, if found continue 16 block zero, img2simg will generate a FILL (zero) 16 chuck (only few byte), instead of DATA chuck. (which is 16*block size). it don't create No-CARE chuck because it don't know if these zero is useful.  

when uboot receive FILL (zero), uboot will write  0 into sdcard or emmc.  that's reason why long time.

zm33y, 2021-10-01:
Ok, thank you for your answers @nxpfrankli! We can close the issue now

closed 2021-10-01 by nxpfrankli