nxp-imx/mfgtools #300: qustion: flash bootloader to sdcard with uuu

qianfan-Zhao, 2021-10-21:
Hi:

I found this scripts in uuu/sd_burn_loader.lst

```
FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev ${sd_dev}
FB: ucmd mmc dev ${sd_dev}
FB: flash bootloader _image
FB: Done
```

And have some quesitions:
1. bootloader should a MBR partition, and how to create this partition by using uuu?
2. RM told that bootloader should position in 1K offset when booting from SD. this is means bootloader partition located at 1K offset?

Thanks

nxpfrankli, 2021-10-21:
uuu -b sd u-boot-with-spl.imx  
your uboot need enable fastboot support

qianfan-Zhao, 2021-10-21:
I am tested with 'mx6ul_14x14_evk_defconfig', fastboot is enabled.  it doesn't work.

nxpfrankli, 2021-10-21:
what's problem your met? Any log from uart?