nxp-imx/mfgtools #304: Unable to copy bz2 file verbative to target

elupus, 2021-11-24:
The automatic detection of bz2 file and appending of /* breaks the ability to copy a bz2 file as is to target. Or let the target decompress it.

uuu_script_recovery 
```
# arguments for uuu script:

# @_flash_update                  | Update image to run

FBK: ucmd echo "Extract update archive"

FBK: acmd tar -C / -xj 2>&1
FBK: ucp _flash_update t:/
FBK: sync
```

Called with
```
 .\uuu.exe -v -b .\uuu_script_recovery updatearchive.tar.bz2
```
Fail since the built in command mangles the argument by rewriting it as updatearchive.tar.bz2/*. 

The automatic decompression also fails in my case, maybe due to sub directories in the archive. But i only end up with a file actually named "*" on the target.

nxpfrankli, 2021-11-28:
auto replace only happen at cmd line with -b option.  If you put filename in script directly, it should work.

elupus, 2021-11-28:
Yes, but those scripts don't support parameters.

nxpfrankli, 2021-12-02:
we can add command option to disable such replacement.  What's good name for such option?

elupus, 2021-12-02:
Why not just assume if the zip file have a trailing /* issue decompression. If not, assume verbatime copy.

nxpfrankli, 2021-12-02:
At beginning, uuu work as what's you said. but I found 

The highest frequency user case is
 
uuu -b emmc_all  rootfs.bz2

Most time,  rootfs.bz2 is http link

uuu -b emmc_all http://xxxxxxx, which is quite long.  It will not convenience to append a * after file.

most people use uboot FB to write whole rootfs image instead extract tar file.  The efficiency of write many small file is quite low. Actually decompress bz2 is quite slow, especially running at target boards.

nxpfrankli, 2021-12-02:
And I think if you remove -xj in tar, it should work. 

Host will extract bz2 file to tar file, just transfer speed become little bit longer. 

Host will use all Core to decompress bz2 parallels.  although increase transfer time, reduce target decompress time (only one core, default),

elupus, 2021-12-07:
Might work. But i find the "automagic" behaviours very confusing.

nxpfrankli, 2021-12-07:
The user also complain /* is confused and not common sense.

nxpfrankli, 2021-12-07:
but sometimes, they also used uncompressed version's rootfs.