nxp-imx/mfgtools #305: HID(W):LIBUSB_ERROR_TIMEOUT with raspberry pi 4 as host

rise-ts, 2021-11-30:
Hello,
I build uuu on raspberry pi 4 an tried to flash u-boot on an imx6ul.

The board is recognized correctly (dmesg):

```
[   84.456977] usb 1-1.4: new high-speed USB device number 4 using xhci_hcd
[   84.557799] usb 1-1.4: New USB device found, idVendor=15a2, idProduct=007d, bcdDevice= 0.01
[   84.557815] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   84.557826] usb 1-1.4: Product: SE Blank 6UL 
[   84.557837] usb 1-1.4: Manufacturer: Freescale SemiConductor Inc 
[   84.561692] hid-generic 0003:15A2:007D.0002: hiddev0,hidraw0: USB HID v1.10 Device [Freescale SemiConductor Inc  SE Blank 6UL ] on usb-0000:01:00.0-1.4/input0
```

This is the script I use:

```
uuu_version 1.2.39

SDP: boot -f ./SPL
# Note: use the *.imX file NOT the *.img file
# Note: be explicit according the target load addr because on some platforms uuu
#       doesn't invoke the correct addr by default
SDPV: write -f ./u-boot.imx -addr 0x877ff400
SDPV: jump -addr 0x877ff400
SDPV: done
```

Output from uuu:

```
sudo /opt/uuu -V /opt/boot.lst 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.165-8-g87a99ff

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDPS:	 MX8ULP		 0x1fc9	 0x014a
	SDPS:	 MX8ULP		 0x1fc9	 0x014b
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001
Wait for Known USB Device Appear...
New USB Device Attached at 1:14
1:14>Start Cmd:SDP: boot -f ./SPL
6400%1:14>Okay (0.049s)
libusb: error [udev_hotplug_event] ignoring udev action bind
New USB Device Attached at 1:14
1:14>Start Cmd:SDPV: write -f ./u-boot.imx
1:14>Fail HID(W):LIBUSB_ERROR_TIMEOUT(1.011s)
```

I use the same script on a ubuntu host and even a raspberry pi 3 with no issue.
What I tried so far:

- different versions of libusb
- different versions of uuu 
- imx_usb_loader
- replace distribution (Raspberry Pi OS (Bullseye) and Ubuntu 21.10). 
- Official snap packet 
- https://github.com/NXPmicro/mfgtools/issues/76

Not luck with this :(
Could it be a problem with the USB driver?

Thank you very much for your help!

lznuaa, 2021-11-30:
1:14>Start Cmd:SDPV: write -f ./u-boot.imx

but script SDPV: write -f ./u-boot.imx **-addr 0x877ff400**

look like missed -addr?   Does SPL hang during download?

rise-ts, 2021-11-30:
Oh sry, looks like this line was not copied. I added it to first post.
No SPL download is working correct.

rise-ts, 2021-12-02:
Closed without a comment? Not very nice :(

nxpfrankli, 2021-12-02:
I think your problem already resolved by add -addr. 
Any misunderstand?

rise-ts, 2021-12-02:
-addr was not copied when I posted the log here, so I added it to my inital post. The problem still persists.

nxpfrankli, 2021-12-02:
Any log from debug uart

rise-ts, 2021-12-03:
That's all I can see:
```
U-Boot SPL 2021.04 (Aug 23 2021 - 14:37:26 +0000)
Trying to boot from USB SDP
SDP: initialize...
SDP: handle requests...
Downloading file of size 445440 to 0x877ff400... 
```
and here the timeout happens.

nxpfrankli, 2021-12-03:
SDP command already sent. Look like spl hang.

rise-ts, 2021-12-06:
But why is it working on a PC or a RasPi 3?

nxpfrankli, 2021-12-06:
I don't know. If want to debug this problem, need begin from spl. If you have usb analyzer, it will be help. if no, need print some log at spl, check all EP0 OUT package.  

Generally, timeout happen caused by device don't ACK host's OUT ping package for quite long time 2s.

rise-ts, 2021-12-16:
The customer has decided not to pursue the error any further and will use a RasPi3. Therefore, I am closing the ticket.

closed 2021-12-16 by rise-ts

stevanradakovic, 2022-02-18:
I'm sorry, is there no plan to support this in the future? Would be nice to have RPi4 support as well, we use it predominantly in our automated testing setup.

nxpfrankli, 2022-02-18:
I have not RP4 to debug this problem.

sousalucascarv, 2023-10-23:
Hello! Any updates on this issue? I'm experiencing the same problem as the author of this issue and have even tried the same potential solutions, but without success so far. My script works on Windows 11 and RPi3, but it doesn't work on RPi4. Is there anything else, aside from what was outlined in the original post, that I could try to fix it?

efecanicoz-atlas, 2023-11-07:
We are also facing this issue while flashing imx7d, it works on windows 11, ubuntu 22.04, and rPi3b but not with rPi4. Is there any debugging guideline to identify issue on our side ?

vj-hansen, 2024-02-02:
FWIW, this should work on the RPi5. I had a similar problem on the RPi4.