nxp-imx/mfgtools #307: Error with QSPI flashing script

danie007, 2021-12-07:
Hi,

I get the following error when executing the script with `uuu` tool:

```bash
Success 0    Failure 1                                                                                                                                                                                            
                                                                                                                                                                                                                   
                                                                                                                                                                                                                   
1:12     3/ 9 [                                      ] FB: ucmd if test ! -n "$fastboot_bytes"; then setenv fastboot_bytes 0x0098718; fi 
```

https://github.com/NXPmicro/mfgtools/blob/64ca58f40faa2ed0efabb391583b31c28e9ba39f/uuu/qspi_burn_loader.lst#L32

- If I replace the above line with `FB: ucmd setenv fastboot_bytes "0x0098718"` the script works as intended. Is there a workaround?
- What the the `$filesize` in the script? Currently I'm using the size of the U-Boot in hex, is that correct?

nxpfrankli, 2021-12-07:
Maybe your uboot version is too old Or have not use our released uboot. 

$filesize is how much byte download by fastboot. 

Early uboot have not $filesize environment, so we added fastboot_byte in nxp's uboot. but upstream uboot already added $filesize environment.

danie007, 2021-12-08:
Hi @nxpfrankli 

Thank you for the comments. We're using the U-Boot version 2021.04 from the [NXP's Yocto hardknott branch](https://source.codeaurora.org/external/imx/imx-manifest/tree/imx-5.10.52-2.1.0.xml?h=imx-linux-hardknott) yet we don't have either `fastboot_bytes` or `filesize` ENV variables set.

[image]

- Do we need to enable some configurations, to have these?

---

CC @tkmozhi

nxpfrankli, 2021-12-08:
Please check after
FB: download -f _image

this environment set after "FB: download -f image",   You can ctrl-C to break fastboot cmd and enter console, 

Or 
FB: download -f image
FB: ucmd echo $filesize

danie007, 2021-12-09:
@nxpfrankli 

Thank you for the clarification. However, I'm still facing the issue with the above mentioned command.
- Could you kindly confirm that the script will work on U-Boot `v2021.04` ?

nxpfrankli, 2021-12-09:
yes, check https://source.denx.de/u-boot/u-boot/-/blob/v2021.04/drivers/fastboot/fb_command.c line 298

danie007, 2021-12-09:
Thanks, now `filesize` is sorted. However, I'm receiving the same error with the default command. I had to change the command as follows,

```bash
FB: ucmd setenv fastboot_bytes $filesize
```

Could you comment on this?

---

CC @tkmozhi

nxpfrankli, 2021-12-09:
FB: ucmd if test ! -n "$fastboot_bytes"; then setenv fastboot_bytes $filesize; fi 

so the condition check is wrong, so not execute "setenv fastboot_bytes $filesize;"

did you enable "test" cmd in your uboot?

danie007, 2021-12-09:
Yes, I tested the `test` command manually and it works as intended.

nxpfrankli, 2021-12-09:
can you check which print if fastboot_bytes is not defined

if test ! -n "$fastboot_bytes"; then echo abc; else echo 123; fi

danie007, 2021-12-10:
Hi @nxpfrankli 

Find the requested command ouput as below:

[image]

It indicate that the `test` command is working and `fastboot_bytes` is not defined. 

---

Interestingly `fastboot_bytes` is set after the `download` command and [the script](https://github.com/NXPmicro/mfgtools/blob/64ca58f40faa2ed0efabb391583b31c28e9ba39f/uuu/qspi_burn_loader.lst) works with the following modification:

```bash
FB: ucmd if test ! -n "$fastboot_bytes"; then setenv fastboot_bytes $filesize; else true; fi
```

I think the problem is the missing `else` statement. The above approach is also used in the script itself.

- Is the `fasboot_bytes` set dynamically on runtime? Can you point a reference for this.
- Shall I raise a PR to close this issue?

---

CC @tkmozhi

PR #309 (2021-12-10): Fixed QSPI flashing script
On U-Boot version `2021.04` the ENV variable `fastboot_bytes` is set dynamically after `download` and this causes the script to stop abryptly. Hence added dummy `else` condition to ensure the proper flow.

The following image shows the `fastboot_bytes` set to `filesize` after the download command:
[image]

closes #307 

---

CC @tkmozhi
closed 2021-12-10 by nxpfrankli