nxp-imx/mfgtools #312: Should correct libusb usage

nmmanx, 2021-12-14:
Hi @nxpfrankli,

Please consider two following problems with libusb usage:

***Problem 1:***
After completely opening a device with `libusb_open()`, we're expected to unreference it and free the list obtained by `libusb_get_device_list()` [(doc).](https://libusb.sourceforge.io/api-1.0/group__libusb__dev.html#gac0fe4b65914c5ed036e6cbec61cb0b97) However, some pieces of code don't follow this rule:

```cpp
int CmdUsbCtx::look_for_match_device(const char *pro)
{
	...
	while (1)
	{
		libusb_device **newlist = nullptr;
		libusb_get_device_list(nullptr, &newlist);
		size_t i = 0;
		libusb_device *dev;

		while ((dev = newlist[i++]) != nullptr)
		{
			...
			ConfigItem *item = get_config()->find(desc.idVendor, desc.idProduct, desc.bcdDevice);
			if (item && item->m_protocol == str_to_upper(pro))
		        {
			        ...
			        int ret;
			        if (ret = open_libusb(dev, &(m_dev)))
				        return ret; // <= Need free newlist
        
			        nt.str = (char*)str.c_str();
			        call_notify(nt);
        
			        return 0;  // <= Need free newlist
		        }
		}

		libusb_free_device_list(newlist, 1);
		...
	}

	return -1;
}
```
The same with `uuu_for_each_devices()`.

***Problem 2:***
I think the `libusb_exit()` calls should be put in appropriate positions. Currently, it's placed only in `uuu_for_each_devices()`. Also, we should be careful with `libusb_exit()` as some detached `std::thread` may still hold references to libusb devices.

Thank you for reading!

nxpfrankli, 2021-12-14:
Thank you for your comments.  

Problem 1： yes, that's problem, should be fixed, otherwise memory leak. 
Problem 2:  Yes, it is wrong.  libusb_exit should be last time.

nxpfrankli, 2021-12-14:
I did quick fix. But it is not good enough. Because daemon mode using ctrl - c to exist,  let me think a good method safely stop thread.

nmmanx, 2021-12-15:
Hi,

If user triggers Ctrl+C (SIGINT), the default handler will terminate the process, and its child threads (whether detached or not) will be gone as well (the same behavior on both Linux & Win). I think handling these signals is the job of applications (uuu.exe) not libuuu. Since the process is gone and resources are claimed by OS, we need not to care about `libusb_exit()` anymore.

As for the problem 2, I mean in the normal execution, application may call multiple libuuu's APIs at anywhere, and in an arbitrary order. So where should we put `libusb_exit()`? In this case, I think we can:

1. Provide another API (e.g `uuu_deinit()`), then applications have to call this function just before exiting from `main()`, or inside a signal handler.
2. Wrap up everything to libuuu APIs, then every libuuu's API that uses libusb need to call `libusb_init()/libusb_exit()` properly, make sure everything is cleaned before returning.

As we spawn a number of threads to execute commands, we should manage them, ensure they are completely finished after `uuu_wait_uuu_finish()` returned. The current implementation will be failed in non-daemon mode if there are two devices running at the same time and one of them finished earlier (depends on usb poll period). The first finished thread will trigger NOTIFY_DONE, and causes `polling_usb()` to return, without waiting for the others.

Thanks!

nxpfrankli, 2021-12-15:
Yes, we can wrap libusb_open/exit to uuu_init and uuu_deinit API.  or using C++ global variable in library to do it automatically. 

If non-daemon,  uuu supposed only support one devices.  If multi-devices connected must be use daemon mode.

closed 2022-01-07 by nmmanx