nxp-imx/mfgtools #314: Corrupt image with larger transfer sizes

tmelin, 2021-12-21:
Tested with master commit 49afb54.
File that triggered error was of size 32509758 bytes.
command was 

`SDPV: write -f image.fit -addr 0x89000000`

File transfer is successful, but sanity checks in U-Boot side (version 2021.10) fails. Error
~~~~
## Checking Image at 89000000 ...
   FIT image found
FIT check error -22
Bad FIT image format!
~~~~

More specifically fdt_check_full() in U-Boot fails when checking FDT_PROP nodes as the propname is garbled. Error code is FDT_ERR_TRUNCATED
Same image can be loaded without errors with previous uuu version in use, 1.2.135. Hence it would initially indicate that the image gets corrupted during the transfer.

nxpfrankli, 2021-12-27:
Can you sent me your fit image?

nxpfrankli, 2021-12-27:
Yes,  I found it. 
We limited size to 32MB because the first fat partition of wic image built by yocto is 32MB.  
We supposed boot image also used for SD boot. 

Let me think how to balance this.

nxpfrankli, 2022-01-04:
Can you try branch b314? https://github.com/NXPmicro/mfgtools/tree/b314

tmelin, 2022-01-05:
Tested with b314 branch commit 06a6cc2 ("Fixed 314 Corrupt image with larger transfer sizes at sdp(s).")
I can confirm that with change transfer is now successful and image is bootable.

(As a side note I occasionally see sha256 checksum for image not matching with expected. That also happens with 1.2.135 when transferring large image but is likely related to some other issue.)

Thanks for fixing this!

nxpfrankli, 2024-12-13:
Are you sure you download to free address of SPL?

fhg0mdv, 2024-12-26:
Actually, I tried with  branch b314? https://github.com/NXPmicro/mfgtools/tree/b314 , the bz2 compressed rootfs, and it is succesfull, but I have a problem with jffs2 file format, is how I can send this file to device, e.g. 

CMD state="Updater" type="push" body="$ flash_erase /dev/mtd%part_rootfs% 0 0">Erasing rootfs partition</CMD>
	 
	 <CMD state="Updater" type="push" body="send" file="files/myfile.rootfs.jffs2" ifdev="MX6ULL">Sending JFFS2 file</CMD>
	 
	 <CMD state="Updater" type="push" body="$ nandwrite -p /dev/mtd4 -p $FILE">Flashing File</CMD> , this is from mfgtool, ucl2 file, how I can do this with uuu tool, in this version(ucl2) , I can send the file but it stops on 100MB, which I described previously, so 
	 1-) I need a way to change the limit for that ucl2(MFGtool2) to upload over 100MB files,
	or 2-) I need a way to upload the jffs2 file to device using uuu tool, so I need a commands for this, 
	
	Can you help about that, kind regards.