nxp-imx/mfgtools #315: 155105 segmentation fault

turboted, 2022-01-14:
I have a segmentation fault  ,log message as below.
environment: aarch64 ubuntu .

uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.191-0-g411f78f

uuu [-d -m -v -V] <bootloader|cmdlists|cmd>

    bootloader  download bootloader to board by usb
    cmdlist     run all commands in cmdlist file
                If it is path, search uuu.auto in dir
                If it is zip, search uuu.auto in zip
    cmd         Run one command, use -H see detail
                example: SDPS: boot -f flash.bin
    -d          Daemon mode, wait for forever.
    -v -V       verbose mode, -V enable libusb error\warning info
    -dry        Dry run mode, check if script or cmd correct 
    -m          USBPATH Only monitor these paths.
                    -m 1:2 -m 1:3

    -t          Timeout second for wait known usb device appeared
    -T          Timeout second for wait next known usb device appeared at stage switch
    -e          set environment variable key=value
    -pp         usb polling period in milliseconds
uuu -s          Enter shell mode. uuu.inputlog record all input commands
                you can use "uuu uuu.inputlog" next time to run all commands

uuu -udev       linux: show udev rule to avoid sudo each time 
uuu -lsusb      List connected know devices
uuu -IgSerNum   Set windows registry to ignore USB serial number for known uuu devices
uuu -h          show general help
uuu -H          show general help and detailed help for commands

uuu [-d -m -v] -b[run] <emmc|emmc_all|fat_write|nand|qspi|sd|sd_all|spl> arg...
	Run Built-in scripts
	emmc	burn boot loader to eMMC boot partition
		arg0: _flash.bin  bootloader
		arg1: _image[Optional]  image burn to emmc, default is the same as bootloader
	emmc_all	burn whole image to eMMC
		arg0: _flash.bin  bootloader, which can extract from wic image
		arg1: _image[Optional]  wic image burn to emmc.
	fat_write	update one file in fat partition, require uboot fastboot running in board
		arg0: _image  image, which cp to fat partition
		arg1: _device  storage device, mmc\sata
		arg2: _partition  fat partition number, like 1:1
		arg3: _filename[Optional]  file name in target fat partition, only support rootdir now
	nand	burn boot loader to NAND flash
		arg0: _flash.bin  bootloader
		arg1: _image[Optional]  image burn to nand, default is the same as bootloader
	qspi	burn boot loader to qspi nor flash
		arg0: _flexspi.bin  bootloader
		arg1: _image[Optional]  image burn to flexspi, default is the same as bootloader
	sd	burn boot loader to sd card
		arg0: _flash.bin  bootloader
		arg1: _image[Optional]  image burn to emmc, default is the same as bootloader
	sd_all	burn whole image to sd card
		arg0: _flash.bin  bootloader, which can extract from wic image
		arg1: _image[Optional]  wic image burn to emmc.
	spl	boot spl and uboot
		arg0: _flash.bin 

uuu -bshow <emmc|emmc_all|fat_write|nand|qspi|sd|sd_all|spl>
	Show built-in script

Enjoy auto [tab] command complete by put below script into /etc/bash_completion.d/uuu
  _uuu_autocomplete()
  {
       COMPREPLY=($(/usr/local/bin/uuu $1 $2 $3))
  }
  complete -o nospace -F _uuu_autocomplete  uuu

[1]    155105 segmentation fault (core dumped)  ./uuu

nxpfrankli, 2022-01-17:
is it related with 

https://github.com/NXPmicro/mfgtools/commit/5397913ad97db422c1d70f314dedff4cb7d976b9

Can you revert above patch and try again?

turboted, 2022-01-18:
libuuu_1.4.191-0-g411f78f already have this patch,so the problem has not been solved

nxpfrankli, 2022-01-18:
How you resolved problem? revert 5397913?

turboted, 2022-01-19:
yes. revert 5397913, delete  class CAutoDeInit .

nxpfrankli, 2022-01-19:
Can you try latest commit (fdead9d170c8585b2949714b14986153cbafb040)?

It should be caused by call libusb_deinit without call libusb_init.