nxp-imx/mfgtools #317: UUU interfacing options other than USB OTG

danie007, 2022-03-09:
Hi @nxpfrankli 

We want to use `UUU` for flasing eMMC of i.MX 6SX based custom board. We don't have the **USB OTG** port available (_yet!_).

### Can we interface the board in `UUU` with any other interface?

Such as,
1. USB Host
2. UART
3. Ethernet

---

CC @tkmozi

nxpfrankli, 2022-03-09:
The key is that ROM only support USB serial download mode. 
If there are already software running at other storage,  the existed uboot tools can already hang USB Host (usb start), UART (rz), Ethernet(tftp).

danie007, 2022-03-11:
@nxpfrankli thank for the answer. ✌🏾 Recently we've come across `SDP` in U-Boot. Is it possible to utilise that (_somehow!?_) in the `UUU` tool.

[doc/imx/misc/sdp.txt](https://source.codeaurora.org/external/imx/uboot-imx/tree/doc/imx/misc/sdp.txt?h=lf_v2021.04)

We've seen the references for `imx_usb` but not the `UUU`.

---

CC @tkmozhi

nxpfrankli, 2022-03-11:
Yes,  you can uuu to talk with uboot with SDP. 

You need write a script 

CFG: SDP -chip uboot -pid <your pid> -vid <your vid>
SDP: .....

If you just do test, you can change uboot's pid/vid is the same as built-in one.