nxp-imx/mfgtools #320: i.MX UUU crashing on Windows PC

JulienJayat, 2022-04-11:
The problem happens with i.MX connected or not-connected to the PC. Older UUU releases and to run cmd as administrator also has been tried. 
In all situations there is a kind of crash/error. The only difference is that older UUU releases print an error and new release print warning in the console, but everything else is the same. From my point of view this looks related to libusb not being able to read USB device ID. But UUU should not crash! 

OS version:

    Edition: Windows 10 Enterprise - all updates installed (Early April 2022).
    Version: 20H2
    OS Build: 19042.1586
    Experience: Windows Feature Experience Pack 120.2212.4170.0
    Laptop: Dell Latitude 5490 with three USB 3.0 ports (tried all of them, where I connected USB directly into the laptop)

Problem description:
using https://github.com/NXPmicro/mfgtools/releases/download/uuu_1.4.193/uuu.exe
[image]
[image]

milanb94, 2022-04-11:
Hello,
I originally reported this problem. This is solved by formatting SSD and installing Windows 10 again.
As far as I'm concerned this topic can be closed.

Thank you,
Milan

milanb94, 2022-04-12:
At the beginning I tried a lot of versions: newer, older,.... and no of them worked. When I got my Windows reinstalled I tried only newest version and it works therefore I am using only that one. (my guess is that all versions will work now)