nxp-imx/mfgtools #325: bz2 images does not work from scripts.

te-johan, 2022-06-07:
when running "FB: flash -raw2sparse all image-imx8mp-bringup.wic.bz2/*"
inside a script UUU fails to locate the file. seems like the ">" magic path the dropped causing it to append the root path twice.

nxpfrankli, 2022-06-07:
We use it daily. what error message?

te-johan, 2022-06-07:
It reports that it cannot find the file `Error: fail open file: >/home/johan/Desktop/deploy/images/imx8mp-bringup/image-imx8mp-bringup.wic.bz2/*`
I start it using `./uuu deploy/images/imx8mp-bringup/uuu.auto` and image-imx8mp-bringup.wic.bz2 is in the same directory. 
`FB: flash -raw2sparse all te-image-imx8mp-bringup.wic.bz2/*` is run inside the uuu.auto script.

By setting a breakpoint at at `get_file_buffer` I can the issue.
`get_file_buffer (filename="image-imx8mp-bringup.wic.bz2/*", async=true) at /home/johan/source/mfgtools/libuuu/buffer.cpp:1165`

`get_file_buffer (filename="/home/johan/Desktop/deploy/images/imx8mp-bringup/image-imx8mp-bringup.wic.bz2", async=true) at /home/johan/source/mfgtools/libuuu/buffer.cpp:1165`

during the second call the path is appended again resulting in `$2 = ">/home/johan/Desktop/deploy/images/imx8mp-bringup//home/johan/Desktop/deploy/images/imx8mp-bringup/image-imx8mp"...
`