nxp-imx/mfgtools #328: Error: can't find ext name in path

mprt, 2022-07-21:
This looks like a regression from uuu_1.4.224 (works) to uuu_1.4.231 (Fails).

I'm using uuu to flash the iMX8 on my Librem5:
https://developer.puri.sm/Librem5/Development_Environment/Phone/Troubleshooting/Reflashing_the_Phone.html

If I connect the device via USB, uuu fails with "Error: can't find ext name in path".

The lst-file looks like this:

```
cat flash_librem5r4.lst 
uuu_version 1.0.1

CFG: SDP: -chip MX8MQ -compatible MX8MQ -vid 0x316d -pid 0x4c05
SDP: boot -f u-boot-librem5.imx
# This command will be run when use SPL
SDPU: delay 1000
SDPU: write -f u-boot-librem5.imx -offset 0x57c00
SDPU: jump
SDPV: delay 1000
SDPV: write -f u-boot-librem5.imx -skipspl
SDPV: jump
# This command will be run when ROM support stream mode
SDPS: boot -f u-boot-librem5.imx
SDPU: delay 1000
#FB: ucmd setenv fastboot_buffer 0x43000000
FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev 0
FB: flash -raw2sparse all librem5r4.img
FB: Done
```

Let me know, if I can halp with debugging

nxpfrankli, 2022-07-21:
Does current working directory is the same as flash_librem5r4.lst? 

uuu flash_librem5r4.lst Or uuu <somepath>\flash_librem5r4.lst

mprt, 2022-07-23:
It's called as
```
subprocess.CalledProcessError: Command '['uuu', '-v', 'tmp/flash_librem5r4.lst']' returned non-zero exit status 255.
```
so the lst file is not in the current working directory.

nxpfrankli, 2022-07-24:
Does (https://github.com/NXPmicro/mfgtools/releases/tag/uuu_1.4.232) fix this problem?

mprt, 2022-07-26:
Yes, I can confirm that 1.4.232 works again!
Thanks!