nxp-imx/mfgtools #330: issue with environment variables on windows

lakabd, 2022-08-29:
Hi,
While experimenting with uuu (in the objectif of replacing an existing procedure that uses the old mfgtool)  I encountered an issue with environment variables definition on windows.

When defining an environment variable for uuu (with -e option), I found out that on windows (in contrary to linux) there is no way to define **empty** environment variables !

for example on Linux the following will work without issues: 
```
$ cat test.conf
uuu_version 1.4.243
SDP:@ boot -f "boot/files/u-boot-mx6@suffix@.imx"
SDP: done

$ uuu -e suffix= test.conf
```
the suffix variable will be replaced by an empty char and the SDP command will succeed (in this way I can decide which file to use by just specifying or not the env variable suffix.. the objective on windows is to automate the definition of the variable in a batch script).

However, on windows, for the same cmd I get the error :
`variable 'suffix' is not defined`

I would like to know if there is a way to have the same behavior as on linux ?

PS: I used windows10

Thank you

nxpfrankli, 2022-08-29:
windows treat suffix= as delete environment suffix. 
uuu can treat un-existed environment as empty at windows platfrom.  

but behavior will be difference when met unexisted environment, windows will treat as empty and linux treat as error.

lakabd, 2022-08-29:
Hi @nxpfrankli , thanks for you reply !
indeed suffix= will delete the env variable for windows. However, if uuu can somehow let me define variables at the cmd line in windows (as -e suffix= without erroring out) and treat them as empty, it can be enough for me. 

Could you please tell me what can I modify to have this behavior ?

lakabd, 2022-08-29:
After some digging, I found out that I can modify the line 776 in libuuu/cmd.cpp like the following:
```
- return {false, {}}
+ return {true, "\0"}
```
and it will give me the behavior I wanted i.e., for non existing environment variables I get a null char !

However, when I compiled the project (uuu.sln) with MSbuild 2017 (version in capture below) I started getting the error below for my uuu script (if I use cmds in the forme `FBK:@ if "@spifw@" != "" then ucp /spi/@spifw@ t:/tmp/fwupdate/@spifw@` where there is multiple environment variables in the same cmd !). **The error appears even without the modification I mentioned above**

[image]
[image]

nxpfrankli, 2022-08-29:
Did you buid 64bit version?  actually, long time not test 32bit version

nxpfrankli, 2022-08-30:
I am try to drop vs2017 support. 
after vs2017, 
<WindowsTargetPlatformVersion>10.0</WindowsTargetPlatformVersion>
can auto selected latest installed.

<PlatformToolset>$(DefaultPlatformToolset)</PlatformToolset>

zstd is not mantained by uuu, which directly use upstream vs2010 project

lakabd, 2022-08-30:
The Readme of the repo says to use VS2017 is it not supported yet ?
I have tried to build with windows10 SDK version 10.0.17134.12 and I'm getting always the same error for my uuu script.

have you tried  to reproduce the issue ? using a cmd in the uuu script file that contains multiple environment variables, such as:
`FBK:@ if "@spifw@" != "" then ucp /spi/@spifw@ t:/tmp/fwupdate/@spifw@`
you don't need a board to test this... the error appears before establishing the connection.

if not, can you please share the setup you are using to build uuu:
 - windows OS version
 - MSbuild tool version (v140, v141,....)
 - SDK version (newer or older than 10.0.16299.0 )
and also what windows SDK version you are using to build zstd? do you have vs2010 installed on your PC ?

nxpfrankli, 2022-08-30:
I am using vs2022. Just update computer!
our CI using vs2019 and vs2022. 

Previous prebuild image built by vs2017.

Here is appveryor config

https://www.appveyor.com/docs/windows-images-software/

lakabd, 2022-08-30:
Thanks for the quick replay @nxpfrankli.

looking at th config of your CI you do have SDK 10.0.16299.0 installed with newer ones also...
I'm clueless, I'll try to change the PC I'm using and see if it makes any difference.

However, can you please try to reproduce the issue on your personal PC ? using a cmd in the uuu script file that contains multiple environment variables, such as:
FBK:@ if "@spifw@" != "" then ucp /spi/@spifw@ t:/tmp/fwupdate/@spifw@
you don't need a board to test this... the error appears before establishing the connection.

nxpfrankli, 2022-08-30:
Yes, I can duplicate your issue.
 
That's because windows treat spifw= as delete environemnt  variable.
linux treat spifw= as add a empty environemnt  variable.

uuu can treat un existed environment variable as empty variable. 

but there will be difference behavior when unexisted variable. 
1 windows think empty.
2. linux report error.

Look like there are always a little bit differenece behavior between windows and linux.

lakabd, 2022-08-30:
I meant the runtime issue I got here.
[image]
for the environment I understood how it works.

nxpfrankli, 2022-08-30:
Sorry, I see.

nxpfrankli, 2022-08-30:
try latest fixed patch

lakabd, 2022-08-30:
I added your modif manually and now it is working like a charm (y), thanks !

And for having the empty environment variables in windows (to have the same behavior as Linux) would you accept to make my modif in main line ? the modif below: (line 776 in libuuu/cmd.cpp)
```
- return {false, {}}
+ return {true, "\0"}
```
I can add a std::cout also to let the user know that the environment variable was not found but it will be used as empty !

nxpfrankli, 2022-08-30:
std::cout will impact cmd UI. 

uuu library use set_last_error_string to report error.  So far there are not warning message notified to caller. 

Maybe we can document this difference at wiki page somewhere.

lakabd, 2022-08-31:
So we can just add the modif as it is without the cout.. the user will eventually end up by seeing that he didn't define the variable (in cas he was welling to define a value for that var).

This at least will guarantee that the behavior of uuu is exactly the same between linux and windows !!! so far, I think this is the first difference between the two operating systems for uuu execution and the modif I gave can fix it.

if you'd like we can add a comment in code saying that we return` { true, "\0" };` to have the same behavior as Linux regarding empty environment variables !

nxpfrankli, 2022-08-31:
you can send pull request

lakabd, 2022-08-31:
pull request submitted !