nxp-imx/mfgtools #331: uuu with google coral dev board, SPL: Unsupported Boot Device 12

synaption, 2022-08-29:
I am trying to get UUU working with the google coral devboard that is based on the imx8mq.  I can get fastboot to work, but this would be a pain for bulk flashing.  I am using buildroot which generates imx8-boot-sd.bin as the uboot binary, verified working in emmc with fastboot.  the error I get with uuu is:

U-Boot SPL 2022.04 (Aug 29 2022 - 13:29:54 -0500)
DDRINFO: start DRAM init
DDRINFO: DRAM rate 3200MTS
DDRINFO:ddrphy calibration done
DDRINFO: ddrmix config done
Normal Boot
SPL: Unsupported Boot Device 12
SPL: failed to boot from all boot devices (err=-6)
### ERROR ### Please RESET the board ###

The output on the host device is as follows:

$ sudo uuu -v -b emmc imx8-boot-sd.bin
uuu (Universal Update Utility) for nxp imx chips -- lib1.4.77

Build in config:
Pctl Chip Vid Pid BcdVersion
==================================================
SDPS: MX8QXP 0x1fc9 0x012f [0x0002..0xffff]
SDPS: MX8QM 0x1fc9 0x0129 [0x0002..0xffff]
SDPS: MX8DXL 0x1fc9 0x0147
SDPS: MX28 0x15a2 0x004f
SDPS: MX815 0x1fc9 0x013e
SDPS: MX865 0x1fc9 0x0146
SDPS: MX8ULP 0x1fc9 0x014a
SDPS: MX8ULP 0x1fc9 0x014b
SDP: MX7D 0x15a2 0x0076
SDP: MX6Q 0x15a2 0x0054
SDP: MX6D 0x15a2 0x0061
SDP: MX6SL 0x15a2 0x0063
SDP: MX6SX 0x15a2 0x0071
SDP: MX6UL 0x15a2 0x007d
SDP: MX6ULL 0x15a2 0x0080
SDP: MX6SLL 0x1fc9 0x0128
SDP: MX7ULP 0x1fc9 0x0126
SDP: MXRT106X 0x1fc9 0x0135
SDP: MX8MM 0x1fc9 0x0134
SDP: MX8MQ 0x1fc9 0x012b
SDPU: SPL 0x0525 0xb4a4 [0x0000..0x04ff]
SDPV: SPL1 0x0525 0xb4a4 [0x0500..0x9998]
SDPU: SPL 0x0525 0xb4a4 [0x9999..0x9999]
SDPU: SPL 0x3016 0x1001 [0x0000..0x04ff]
SDPV: SPL1 0x3016 0x1001 [0x0500..0x9998]
FBK: 0x066f 0x9afe
FBK: 0x066f 0x9bff
FB: 0x0525 0xa4a5
FB: 0x18d1 0x0d02
FB: 0x3016 0x0001

Run built-in script:

uuu_version 1.2.39

# @imx8-boot-sd.bin | bootloader
# @imx8-boot-sd.bin [imx8-boot-sd.bin] | image burn to emmc, default is the same as bootloader

# This command will be run when i.MX6/7 i.MX8MM, i.MX8MQ
SDP: boot -f imx8-boot-sd.bin

# This command will be run when ROM support stream mode
# i.MX8QXP, i.MX8QM
SDPS: boot -f imx8-boot-sd.bin

# These commands will be run when use SPL and will be skipped if no spl
# SDPU will be deprecated. please use SDPV instead of SDPU
# {
SDPU: delay 1000
SDPU: write -f imx8-boot-sd.bin -offset 0x57c00
SDPU: jump
# }

# These commands will be run when use SPL and will be skipped if no spl
# if (SPL support SDPV)
# {
SDPV: delay 1000
SDPV: write -f imx8-boot-sd.bin -skipspl
SDPV: jump
# }

FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev ${emmc_dev}
FB: ucmd mmc dev ${emmc_dev}
FB: flash bootloader imx8-boot-sd.bin
FB: ucmd if env exists emmc_ack; then ; else setenv emmc_ack 0; fi;
FB: ucmd mmc partconf ${emmc_dev} ${emmc_ack} 1 0
FB: Done

Wait for Known USB Device Appear...
New USB Device Attached at 1:141
1:141>Start Cmd:SDP: boot -f imx8-boot-sd.bin
6400%1:141>Okay (3.066s)

These are the lines I have added currently to my uboot config

CONFIG_CMD_FASTBOOT=y
CONFIG_USB_FUNCTION_FASTBOOT=y
CONFIG_USB_GADGET=y
CONFIG_USB_GADGET_DOWNLOAD=y
CONFIG_USB_GADGET_MANUFACTURER="FSL"
CONFIG_USB_GADGET_VENDOR_NUM=0x0525
CONFIG_USB_GADGET_PRODUCT_NUM=0xa4a5
CONFIG_USB_DWC3=y #I have tried CONFIG_USB_CDNS3 and CONFIG_CI_UDC
CONFIG_FSL_FASTBOOT=y
CONFIG_FASTBOOT=y
CONFIG_FASTBOOT_BUF_ADDR=0x40480000
CONFIG_FASTBOOT_BUF_SIZE=0x40000000
CONFIG_FASTBOOT_FLASH=y
CONFIG_FASTBOOT_FLASH_MMC_DEV=1
CONFIG_EFI_PARTITION=y
##CONFIG_ANDROID_BOOT_IMAGE=y#it wont compile with this line included.  
# SPL enalbe
CONFIG_SPL_USB_HOST_SUPPORT=y
CONFIG_SPL_USB_GADGET_SUPPORT=y
CONFIG_SPL_USB_SDP_SUPPORT=y
CONFIG_SDP_LOADADDR=0x40400000
CONFIG_USB_TCPC=n #I have tried with and without TCPC
CONFIG_SPL_SHOW_ERRORS=y

 

CONFIG_ANDROID_BOOT_IMAGE=y causes a compile error.  I'm not sure if that matters.

synaption, 2022-08-30:
Thanks nxpfrankli.  What version of uboot are you reverencing?  I do not see this in mainline uboot or imx uboot's board/freescale/imx8mp_evk/spl.c.  When I add this to common/spl/spl.c, I get a build error.  If I understand correctly, I think I need to add something to spl.h as well?
```
common/spl/spl.c:42:32: warning: ‘enum boot_device’ declared inside parameter list will not be visible outside of this definition or declaration
   42 | int spl_board_boot_device(enum boot_device boot_dev_spl)
      |                                ^~~~~~~~~~~
common/spl/spl.c:42:44: error: parameter 1 (‘boot_dev_spl’) has incomplete type
   42 | int spl_board_boot_device(enum boot_device boot_dev_spl)
      |                           ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
common/spl/spl.c: In function ‘spl_board_boot_device’:
common/spl/spl.c:48:6: error: ‘SD1_BOOT’ undeclared (first use in this function)
   48 | case SD1_BOOT:
      |      ^~~~~~~~
common/spl/spl.c:48:6: note: each undeclared identifier is reported only once for each function it appears in
common/spl/spl.c:49:6: error: ‘MMC1_BOOT’ undeclared (first use in this function)
   49 | case MMC1_BOOT:
      |      ^~~~~~~~~
common/spl/spl.c:50:6: error: ‘SD2_BOOT’ undeclared (first use in this function)
   50 | case SD2_BOOT:
      |      ^~~~~~~~
common/spl/spl.c:51:6: error: ‘MMC2_BOOT’ undeclared (first use in this function)
   51 | case MMC2_BOOT:
      |      ^~~~~~~~~
common/spl/spl.c:53:6: error: ‘SD3_BOOT’ undeclared (first use in this function)
   53 | case SD3_BOOT:
      |      ^~~~~~~~
common/spl/spl.c:54:6: error: ‘MMC3_BOOT’ undeclared (first use in this function)
   54 | case MMC3_BOOT:
      |      ^~~~~~~~~
common/spl/spl.c:56:6: error: ‘QSPI_BOOT’ undeclared (first use in this function)
   56 | case QSPI_BOOT:
      |      ^~~~~~~~~
common/spl/spl.c:58:6: error: ‘NAND_BOOT’ undeclared (first use in this function)
   58 | case NAND_BOOT:
      |      ^~~~~~~~~
common/spl/spl.c:60:6: error: ‘USB_BOOT’ undeclared (first use in this function)
   60 | case USB_BOOT:
      |      ^~~~~~~~
common/spl/spl.c:66:1: warning: control reaches end of non-void function [-Wreturn-type]
   66 | }
      | ^
make[4]: *** [scripts/Makefile.build:254: spl/common/spl/spl.o] Error 1
```

nxpfrankli, 2022-08-30:
https://github.com/nxp-imx/uboot-imx/blob/lf_v2022.04/board/freescale/imx8mp_evk/spl.c
line 33

synaption, 2022-08-31:
imx8mp! not imx8mq.  wrong board.  Anyway.  I added ```#include <asm/mach-imx/boot_mode.h>``` and that allowed the code to compile, but this did not change my error message.  I even tried to force the function to ```return BOOT_DEVICE_BOARD;```  This caused the output of ```spl_loader_name()``` to change, so I can tell ```spl_board_boot_device()``` is being called, but ```spl_ll_find_loader()``` still returns 0.

nxpfrankli, 2022-08-31:
Here is major for UUU host side issues.  You'd better post uboot issue in nxp community.
https://community.nxp.com/.  There are more uboot expert to help you. or post into uboot's maillist.

synaption, 2022-09-01:
Is there any chance I will have better luck with a winusb based transfer?  How would I go about doing that.

nxpfrankli, 2022-09-01:
You have to pass this phase,  let SPL receive uboot firstly by SDP. 

you can check common/spl/spl_sdp.c, make sure 
SPL_LOAD_IMAGE_METHOD("USB SDP", 0, BOOT_DEVICE_BOARD, spl_sdp_load_image);

synaption, 2022-09-03:
NEW ERROR! exciting.  I added a lot of include/config/imx8mq_evk.h to include/config/imx8mq_phanbell.h, as well as adding a lot from board/freescale/imx8mq_phanbell/imx8mq_evk.c to board/google/imx8mq_phanbell/imx8mq_phanbell.c.  It seems like everything that should compile does.  Now the process hangs at the following:

```
U-Boot SPL 2022.04 (Sep 03 2022 - 02:34:18 -0500)
DDRINFO: start DRAM init
DDRINFO: DRAM rate 3200MTS
DDRINFO:ddrphy calibration done
DDRINFO: ddrmix config done
Normal Boot
Trying to boot from USB SDP
SDP: initialize...
SDP: handle requests...
```

synaption, 2022-09-03:
```lsusb``` shows the device as ``Bus 001 Device 093: ID 0525:0151 Netchip Technology, Inc. USB download gadget``` but ```0525:0151``` does not seme to corispont to anything in this chart.  I'm not sure if this is the problem or how to fix it.   
```
Build in config:
        Pctl     Chip            Vid     Pid     BcdVersion
        ==================================================
        SDPS:    MX8QXP          0x1fc9  0x012f  [0x0002..0xffff]
        SDPS:    MX8QM           0x1fc9  0x0129  [0x0002..0xffff]
        SDPS:    MX8DXL          0x1fc9  0x0147
        SDPS:    MX28            0x15a2  0x004f
        SDPS:    MX815           0x1fc9  0x013e
        SDPS:    MX865           0x1fc9  0x0146
        SDPS:    MX8ULP          0x1fc9  0x014a
        SDPS:    MX8ULP          0x1fc9  0x014b
        SDP:     MX7D            0x15a2  0x0076
        SDP:     MX6Q            0x15a2  0x0054
        SDP:     MX6D            0x15a2  0x0061
        SDP:     MX6SL           0x15a2  0x0063
        SDP:     MX6SX           0x15a2  0x0071
        SDP:     MX6UL           0x15a2  0x007d
        SDP:     MX6ULL          0x15a2  0x0080
        SDP:     MX6SLL          0x1fc9  0x0128
        SDP:     MX7ULP          0x1fc9  0x0126
        SDP:     MXRT106X        0x1fc9  0x0135
        SDP:     MX8MM           0x1fc9  0x0134
        SDP:     MX8MQ           0x1fc9  0x012b
        SDPU:    SPL             0x0525  0xb4a4  [0x0000..0x04ff]
        SDPV:    SPL1            0x0525  0xb4a4  [0x0500..0x9998]
        SDPU:    SPL             0x0525  0xb4a4  [0x9999..0x9999]
        SDPU:    SPL             0x3016  0x1001  [0x0000..0x04ff]
        SDPV:    SPL1            0x3016  0x1001  [0x0500..0x9998]
        FBK:                     0x066f  0x9afe
        FBK:                     0x066f  0x9bff
        FB:                      0x0525  0xa4a5
        FB:                      0x18d1  0x0d02
        FB:                      0x3016  0x0001
```

lznuaa, 2022-09-06:
There are config to control USB VID and PID. You can reuse NXP one.

CONFIG_USB_GADGET_VENDOR_NUM=0x1fc9
CONFIG_USB_GADGET_PRODUCT_NUM=0x0152

I remember SPL Production_Number is CONFIG_USB_GADGET_PRODUCT_NUM  - 1
 
Suppose your uuu is quit old.

0x0525  0xb4a4   actually is wrong PID/VID, which already register. 

cfg only used for in uuu script. System will forget it after run uuu cfg: FB: ....

synaption, 2022-09-13:
With A LOT of hacking, I have been able to make something work.  CONFIG_USB_GADGET_VENDOR_NUM seemed to effect the right value, but CONFIG_USB_GADGET_PRODUCT_NUM did not do anything unless I changed it directly in the c code.  Ultimately what I am able to do is this:
```
sudo uuu SDP: boot -f SDPVimx8-boot-sd.bin
sleep 1
sudo uuu SDPV: write -f mendel-imx8-boot-sd.bin -skipspl
sudo uuu SDPV: jump
# ums 0 mmc 0 # on coral
sudo bmaptool copy --bmap sdcard.img.bmap sdcard.img.xz /dev/sdb
# fastboot 0 # on coral
fastboot flash bootloader0 imx8-boot-sd.bin
```
So I basically used my uboot to start SDPV, then I load the original mendel linux uboot to use fastboot and ums.  The last problem I am having is that I would rather run fastboot and ums with ```uuu FB: ucmd``` but the mendel uboot automatically boots into fastboot, and I need to use the serial terminal to press ctrl-c.  Is there a way to press ctrl-c with ucmd?

nxpfrankli, 2022-09-13:
fb: continue

closed 2022-09-30 by synaption