nxp-imx/mfgtools #334: "HID(R):LIBUSB_ERROR_TIMEOUT" error when trying to flash bootloader to iMX8MQ EVK board

zlhk100, 2022-10-18:
Constantly failing with "HID(R):LIBUSB_ERROR_TIMEOUT" when using UUU to flash eMMC.

I'm using ubuntu guest on Apple MacBook with M1 SoC.  Please see the screenshot of failing messages.

Between these two commands,  I enabled "SE blank M859" USB ID and "USB mass storage" for ubuntu guest in the background.

[image]

Would much appreciate your support!

Thanks,

nxpfrankli, 2022-10-18:
It looks like SPL dead.

LukasUn, 2022-11-19:
I think I have a related problem. We also get a LIBUSB_ERROR_TIMEOUT.

The UUU Tool does not consistently exit at a fixed step and many of the commands of the "uuu.lst" work fine until the libusberror eventually appears. Hence, initial steps like loading the bootloader or the fsl initramfs work fine. We found that executing commands e.g., with "FSKB: ucmd" are the last ones before the error occurs.

Here is an example output from UUU with an error:
[...]
The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.
2:19>Okay (0.314s)
2:19>Start Cmd:FBK: ucmd sleep 1
2:19>Fail Bulk(R):LIBUSB_ERROR_TIMEOUT(2.008s)

[uuu.lst.txt](https://github.com/NXPmicro/mfgtools/files/10045925/uuu.lst.txt)

nxpfrankli, 2022-11-20:
I have not seen "ucmd sleep 1" in your script

LukasUn, 2022-11-24:
Sorry my bad. We created another script where we added lots of **echos** and **sleeps** to trigger the LIBUSB_ERROR_TIMEOUT every time we run the script. The output from above is from this altered script.
But figured that this would rather add another error than make the error more clear. But nevertheless I added this script here.
[uuu_echo_sleep.lst.txt](https://github.com/NXPmicro/mfgtools/files/10080279/uuu_echo_sleep.lst.txt)