nxp-imx/mfgtools #338: Next release date?

gruizdigi, 2022-11-14:
Hi @nxpfrankli ,

We are soon to do a release that requires one of the latest changes in UUU. I have already a working, manually built, UUU binary but I was wondering if you have an approximate date for the next release in this repo?

Thanks in advance.

nxpfrankli, 2022-11-14:
Which one do you need?  1.5.4?

gruizdigi, 2022-11-15:
The one we need is at least 1.5.6.

nxpfrankli, 2022-11-15:
I create a pre release https://github.com/NXPmicro/mfgtools/releases/tag/uuu_1.5.11

I will remove pre-release take after some month.

gruizdigi, 2022-11-16:
Many thanks @nxpfrankli .

When you remove the pre-release, will there be an official release to replace it?

nxpfrankli, 2022-11-16:
If not problem, I just remove pre-release tag and update release notes.  
If there are problem, new pre-release will be created.

gruizdigi, 2022-11-16:
Great, that will work fine for us. Thank you!

closed 2022-11-16 by gruizdigi