nxp-imx/mfgtools #341: Bulk(W):LIBUSB_ERROR_TIMEOUT when device is in FB mode [imx6ull]

mohamed-aouadhi, 2023-01-13:
Hello, 

We are using a custom board based on imx6ull with uboot installed, we use FB mode of uboot to flash the rootfs image using `uuu` 
```
sudo uuu -v -b emmc_all ${uboot_image} ${rootfs}
```
tool but from time to time we get this error after any command (here after setting a env variable):
```
uuu (Universal Update Utility) for nxp imx chips -- lib1.4.193

Success 0    Failure 1                                                                       
                                                                                              
                                                                                              
1:1      1/ 1 [Bulk(W):LIBUSB_ERROR_TIMEOUT          ] FB: ucmd setenv fdt_addr 0x83000000    

Error: Bulk(W):LIBUSB_ERROR_TIMEOUT
```
Even when we reboot the device, the problem persists, and the only way that seems to work is to intercept the uboot interface (from a UART interface) upon boot and to put it manually in FB mode.

u-boot version (nxp port): lf_2021.04
uuu version: uuu (Universal Update Utility) for nxp imx chips -- lib1.4.193

nxpfrankli, 2023-01-13:
Did you change uuu? 
I have not see  command "FB: ucmd setenv fdt_addr 0x83000000" in our build in script

mohamed-aouadhi, 2023-01-16:
No I didn't change uuu, the log that I put was from a custom script of ours, it's just an example to show that any command involving uuu after that point results in the timeout.

mhadavan, 2023-09-27:
Any updates on this issue?
I have similar issues right now for some boards.
[image]
where loadtftpimage is a custom script