nxp-imx/mfgtools #343: Output from u-boot shell commands via uuu

iandulchinos-cpi, 2023-01-23:
Hello,

We have a custom board using an imx8 processor with u-boot as the bootloader. We interface with the board using `uuu`, and I am looking for a way to get output on the host machine from commands ran in the u-boot shell, specifically `FB` commands. When the board is booted into a linux-based OS, I am able to accomplish this with `-v` when sending `FBK` commands, e.g. `uuu -v FBK: ucmd ifconfig`. I am wondering if this should be expected to work in u-boot as well.

u-boot forked from version v2018.07
Tried on `uuu` version 1.4.139 and 1.5.11

Thanks.

nxpfrankli, 2023-01-23:
Uboot still not support this yet.

iandulchinos-cpi, 2023-01-23:
Alright, thanks. Do you know of any timeline or plan to support this?

nxpfrankli, 2023-01-24:
Sorry, Our uboot support it. I added it at 2019.  totally forget.

commit 264d5ebf365f493ea76d5adf632ac73264265996
Author: Frank Li <Frank.Li@nxp.com>
Date:   Fri Dec 13 14:36:01 2019 -0600

    MLK-23124 fastboot: support duplicate uart console output to uuu
    
    to enable it, in uuu script add below command
    fb: ucmd setenv stdout serial,fastboot
    uuu add -v opition.
    output will show in uuu console
    
    Signed-off-by: Frank Li <Frank.Li@nxp.com>
    (cherry picked from commit a5e86802515221cdfa4b6d82e697cf9dc0d29a98)
    (cherry picked from commit e4bcf17c0c7eaf8365435927be43577a98deed78)
    (cherry picked from commit 1632685a5141001ac2d951cf1ed97e8f45ccaa44)
    (cherry picked from commit d5b90b97cbbf4943ab0752295b9b8c68198d47a8)

iandulchinos-cpi, 2023-01-25:
Great, thanks! Will work on patching this in.

closed 2023-01-25 by iandulchinos-cpi