nxp-imx/mfgtools #344: Frozen when flashing the image for IMX6DL

MihanLee, 2023-01-31:
Hi 

I tried to flash the latest image V5.15.32 for the board based on IMX6U7CVM08AC, using Ubuntu 22.04 with the *.wic.bz2 file

But It's frozen, here is the message in verbose mode:
`vmuser@vmuser-virtual-machine:/mnt/hgfs/CODE$ uuu -v -b emmc_all u-boot-tagged.imx imx-image-full-imx6dlsabresd-20230131030147.rootfs.wic.bz2 
uuu (Universal Update Utility) for nxp imx chips -- lib1.4.193

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDPS:	 MX8ULP		 0x1fc9	 0x014a
	SDPS:	 MX8ULP		 0x1fc9	 0x014b
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPV:	 SPL1		 0x1fc9	 0x0151	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FBK:	 		 0x1fc9	 0x0153
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001
	FB:	 		 0x1fc9	 0x0152

Run built-in script:
 
uuu_version 1.4.149

# @_flash.bin            | bootloader, which can extract from wic image
# @_image   [_flash.bin] | wic image burn to emmc.

# This command will be run when i.MX6/7 i.MX8MM, i.MX8MQ
SDP: boot -f u-boot-tagged.imx -scanlimited 0x800000

# This command will be run when ROM support stream mode
# i.MX8QXP, i.MX8QM
SDPS: boot -scanterm -f u-boot-tagged.imx -scanlimited 0x800000

# These commands will be run when use SPL and will be skipped if no spl
# SDPU will be deprecated. please use SDPV instead of SDPU
# {
SDPU: delay 1000
SDPU: write -f u-boot-tagged.imx -offset 0x57c00
SDPU: jump -scanlimited 0x800000
# }

# These commands will be run when use SPL and will be skipped if no spl
# if (SPL support SDPV)
# {
SDPV: delay 1000
SDPV: write -f u-boot-tagged.imx -skipspl -scanterm -scanlimited 0x800000
SDPV: jump -scanlimited 0x800000
# }

FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev ${emmc_dev}
FB: ucmd mmc dev ${emmc_dev}
FB: flash -raw2sparse all imx-image-full-imx6dlsabresd-20230131030147.rootfs.wic.bz2/*
FB: flash -scanterm -scanlimited 0x800000 bootloader u-boot-tagged.imx
FB: ucmd if env exists emmc_ack; then ; else setenv emmc_ack 0; fi;
FB: ucmd mmc partconf ${emmc_dev} ${emmc_ack} 1 0
FB: done

Wait for Known USB Device Appear...
New USB Device Attached at 3:2
3:2>Start Cmd:SDP: boot -f u-boot-tagged.imx -scanlimited 0x800000
100%3:2>Okay (3.501s)
`

What should I do for it?

nxpfrankli, 2023-01-31:
look uboot have not boot at boards, can you check serial port output?

MihanLee, 2023-02-01:
But I flash the old image (4.1.15) with mfgtool is okay. is there some differents between the latest image (5.15.32) and the old one e.g. address?

MihanLee, 2023-02-01:
I found the size of old u-boot (4.1.15) is about 367KB, and the latest one is 624KB.

MihanLee, 2023-02-02:
I tried to make the SD card as following the guide from NXP to boot, but the unit didn't boot with this image from SD card, which can boot with the old image.
` bzcat ...wic.bz2 || sudo dd of=/dev/sd.. bs=1M conv=fsync `

OTOH,  `uuu -b emmc <old_uboot> <new_uboot>` is also frozen on the same place.

MihanLee, 2023-02-02:
I found some output from serial port using uuu command as you said on Windows.
```
U-Boot 2014.04-gd8ae003-dirty (Oct 26 2017 - 17:19:36)
                                                      
CPU:   Freescale i.MX6DL rev1.3 at 792 MHz
CPU:   Temperature 57 C, calibration data: 0x59751f69
Reset cause: POR                                     
Board: MX6-SabreSD
I2C:   ready      
DRAM:  512 MiB
MMC:   FSL_SDHC: 0, FSL_SDHC: 1, FSL_SDHC: 2
*** Warning - bad CRC, using default environment
                                                
No panel detected: default to Hannstar-XGA
Display: Hannstar-XGA (1024x768)          
In:    serial                   
Out:   serial
Err:   serial
DS2460'S ID is:  0x3c 0x9e 0xc2 0x81 0x00 0x00 0x00 0x44
normol mode!                                            
mmc2(part 0) is current device
Net:   Read MAC from ds2460 : 00:14:97:2e:88:fa
Phy not found                                  
PHY reset timed out
FEC [PRIME]        
Boot from USB for mfgtools
Use default environment for mfgtools
Run bootcmd_mfg: run mfgtool_args;bootz ${loadaddr} ${initrd_addr} ${fdt_addr};
Hit any key to stop autoboot:  0                                               
Bad Linux ARM zImage magic!
```

MihanLee, 2023-02-02:
Ok, how to debug uboot in details?

nxpfrankli, 2023-02-02:
you can ctrl-c, 
then enter console, run fastboot 0

If it works. It should failure auto run fastboot by detect boot from USB. 
if it is not works, it should usb related function does work.

MihanLee, 2023-02-03:
`run fastboot 0` or `fastboot 0` doesn't work on the console.
```
=> run fastboot 0                                   
## Error: "fastboot" not defined
=> fastboot 0 
Unknown command 'fastboot' - try 'help'
=>
```

MihanLee, 2023-02-03:
I also tried to flash the new uboot by mfgtool, it could be finished but the board couldn't boot.

nxpfrankli, 2023-02-03:
Your uboot missed some configuration. https://github.com/NXPmicro/mfgtools/wiki/uboot-config-requirement

MihanLee, 2023-02-06:
Sorry for late.
I added the configuration following the website to `.config` file, but the uboot v2016.03 doesn't seem to have this `fastboot` function.
but I saw something about `SPL` but there is something different. It would be good if you could check it for me. The attach is the `.config` file after `make mx6dlsabresd_defconfig`.
```
#
# Boot images
#
CONFIG_SUPPORT_SPL=y
# CONFIG_SPL is not set
# CONFIG_FIT is not set
CONFIG_SYS_EXTRA_OPTIONS="IMX_CONFIG=board/freescale/mx6sabresd/mx6dlsabresd.cfg,MX6DL,SYS_USE_SPINOR"
```
[config.txt](https://github.com/nxp-imx/mfgtools/files/10661718/config.txt)