nxp-imx/mfgtools #346: when compile based on Raspberry Pi 3b+(ARMv7 Processor rev 4 (v7l)), it shows error

augustine829, 2023-02-10:
pi@CPUTESTER:~/MFGTOOL_WORK/mfgtools $ cmake . && make
CMake Warning (dev) at libuuu/CMakeLists.txt:59:
  Syntax Warning in cmake code at column 159

  Argument not separated from preceding token by whitespace.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/MFGTOOL_WORK/mfgtools
[  2%] Generating gitversion.h
build not in appveyor
Scanning dependencies of target uuc_s
[  5%] Building CXX object libuuu/CMakeFiles/uuc_s.dir/buffer.cpp.o
/home/pi/MFGTOOL_WORK/mfgtools/libuuu/buffer.cpp: In member function 'virtual int ZstdStream::decompress()':
/home/pi/MFGTOOL_WORK/mfgtools/libuuu/buffer.cpp:607:59: error: cannot convert 'ZSTD_DCtx* {aka ZSTD_DCtx_s*}' to 'ZSTD_DStream* {aka ZSTD_DStream_s*}' for argument '1' to 'size_t ZSTD_decompressStream(ZSTD_DStream*, ZSTD_outBuffer*, ZSTD_inBuffer*)'
   return ZSTD_decompressStream(m_dctx, &m_output, &m_input);
                                                           ^
In file included from /home/pi/MFGTOOL_WORK/mfgtools/libuuu/buffer.cpp:48:0:
/usr/include/zstd.h:131:16: note: class type 'ZSTD_DCtx {aka ZSTD_DCtx_s}' is incomplete
 typedef struct ZSTD_DCtx_s ZSTD_DCtx;
                ^~~~~~~~~~~
/home/pi/MFGTOOL_WORK/mfgtools/libuuu/buffer.cpp: In member function 'virtual size_t ZstdStream::decompress_size(const string&)':
/home/pi/MFGTOOL_WORK/mfgtools/libuuu/buffer.cpp:626:67: error: 'ZSTD_getFrameContentSize' was not declared in this scope
   size_t decompressed_sz = ZSTD_getFrameContentSize(pb->data(), sz);
                                                                   ^
libuuu/CMakeFiles/uuc_s.dir/build.make:91: recipe for target 'libuuu/CMakeFiles/uuc_s.dir/buffer.cpp.o' failed
make[2]: *** [libuuu/CMakeFiles/uuc_s.dir/buffer.cpp.o] Error 1
CMakeFiles/Makefile2:85: recipe for target 'libuuu/CMakeFiles/uuc_s.dir/all' failed
make[1]: *** [libuuu/CMakeFiles/uuc_s.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

nxpfrankli, 2023-02-11:
did you install zstd library
sudo apt-get install libzstd-dev?

or your zstd lib too old

augustine829, 2023-02-12:
Thanks, I have a try.

augustine829, 2023-02-12:
I have installed the zstd lib, but it still can not compile showing the same error

augustine829, 2023-02-12:
Reading package lists... Done
Building dependency tree       
Reading state information... Done
cmake is already the newest version (3.7.2-1).
g++ is already the newest version (4:6.3.0-4).
libbz2-dev is already the newest version (1.0.6-8.1).
libssl-dev is already the newest version (1.1.0l-1~deb9u6).
libusb-1.0-0-dev is already the newest version (2:1.0.21-1).
**libzstd-dev is already the newest version (1.1.2-1+deb9u1).**
pkg-config is already the newest version (0.29-4).
0 upgraded, 0 newly installed, 0 to remove and 356 not upgraded.

augustine829, 2023-02-12:
do you know how to upgrade zstd-dev into 1.3?

nxpfrankli, 2023-02-13:
git clone https://github.com/facebook/zstd

make install

augustine829, 2023-02-13:
Thanks so much for your information. I built the uuu.pi, it can use for imx6sx, but it can not use for imx7d, do you know why?

nxpfrankli, 2023-02-13:
what's problem?

augustine829, 2023-02-13:
USB Device: MX6SX,SDP,1:15
USB Device: MX6SX,SDP,1:15
reboot_uic, prot=SDP, full=False
uicLoad - reboot_uic
model=Solo
run: tools/uuu.pi -m 1:15 -t 60  firmware/uic/u-boot-solo.imx
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.21-3-gcbff795

Success 1    Failure 0     at path 1:15                                                                                      
                                                                                                                              
                                                                                                                              
1:15     2/ 2 [Done                                  ] SDP: done                                                              

USB Device: Not Detected, chip=MX6SX, state=FB
USB Device: MX6SX,FB,1:15
USB Device: MX6SX,FB,1:15
CE Load Images for imx7d-p2 Board
PWA:        P4
Board:      imx7d-p2
Board SN#:  aaaaaaaaa
Fixture:    1
USB Device: MX7D,SDP,1:131
USB Device: MX7D,SDP,1:131
reboot_ce, prot=SDP, full=False
run: tools/uuu.pi -m 1:131 -t 60  firmware/ce/u-boot.imx
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.21-3-gcbff795

Success 1    Failure 0     at path 1:131                                                                                     
                                                                                                                              
                                                                                                                              
1:131    2/ 2 [Done                                  ] SDP: done                                                              

USB Device MX7D: USB State: SDP
USB Device MX7D: USB State: SDP
USB Device MX7D: USB State: SDP
USB Device MX7D: USB State: SDP
USB Device MX7D: USB State: SDP
USB Device MX7D: USB State: SDP
USB Device MX7D: USB State: SDP
USB Device MX7D: USB State: SDP
USB Device MX7D: USB State: SDP
USB Device MX7D: USB State: SDP
USB Device MX7D: USB State: SDP
USB Device: Not Detected, chip=MX7D, state=FB
USB Device MX7D: USB State: SDP
USB Device MX7D: USB State: SDP
USB Device MX7D: USB State: SDP
UIC:UIC_CE_PROBE: 1, mx6=1, mx7=0, mx6val=MX6SX,FB,1:15, mx7val=Error: CE SDP Boot Failed!
Check Boot Mode, USB Connection, Board Power,

augustine829, 2023-02-13:
The issue only happens in imx7d, do you know what i need to configure ?
thanks so much

nxpfrankli, 2023-02-13:
UUU have not report error.  it was success.

augustine829, 2023-02-13:
do you know what we need to configure bootload or others ? thanks

augustine829, 2023-02-14:
I have a try to load uboot images using uuu FB or FBK mode, it can load uboot images. But when using SDP, it can not load uboot images. do you know what's the configure of USB or Others?

nxpfrankli, 2023-02-14:
if you want to use FB,  you download uboot by sdp firstly, then ctrl-c at uboot console,  then run `fastboot 0` command. 
if failure, uboot some configuration missed. please refer: 

https://github.com/nxp-imx/mfgtools/wiki/uboot-config-requirement

augustine829, 2023-02-17:
Hi, I have now updated them in imx6, but imx7d can not. do I need to update libusb or stuff?

nxpfrankli, 2023-02-17:
I don't think so.  host side is the same

augustine829, 2023-02-17:
I have another concern about uuu images size. I built the uuu, it's only 787360, but download uuu is 5.7MB. do you know why?

nxpfrankli, 2023-02-17:
prebuild image use many static link to reduce dependence, everyone can use directly.
if you build locally,  dynmatic link used.

augustine829, 2023-02-17:
Thanks for your reply. If I build at locally, what do these dynamical libs require?

nxpfrankli, 2023-02-17:
libusb-1.0-0-dev libbz2-dev libzstd-dev  libssl-dev,   if you can pass built, all dynamtical lib already installed.

augustine829, 2023-02-17:
thanks, is that the latest release based on latest libusb?

augustine829, 2023-02-18:
do you know how to build uuu with static libusb?

nxpfrankli, 2023-02-18:
try,  cmake -D 'STATIC=1', not sure if libusb is in static library list

augustine829, 2023-02-20:
Hi I am now basing on uboot(2018.03) release, what's the version of uboot release referenced on your configure?
and kernel?

augustine829, 2023-02-21:
Hi, what's the uboot version based on the documents uboot configure?