nxp-imx/mfgtools #353: Please provide aarch64 binary for MacOS (aka Apple Silicon)

zyga, 2023-03-08:
M1 and M2 Macs are slowly replacing their existing x86 cousins. CI should be able to build binaries for aarch64.

nxpfrankli, 2023-03-08:
I will add it if appveyor support it.

ricardosalveti, 2023-03-20:
@nxpfrankli were you able to check if there is a way to build for arm64 on appveyor? We had a user looking for uuu_mac on arm64 and found this issue here. 

Looking at https://ci.appveyor.com/project/nxpfrankli/mfgtools-kvqcg there is no recent build to know if it was changed or not.

Thanks!

LYC584520, 2023-08-19:
More people have the same requirements, looking forward for that.

nxpfrankli, 2023-08-21:
Understand, wait for github action support M2 docker.

mwasilew, 2024-04-08:
This already works. I had it running on my fork. This is the change that sends the build to apple silicone:
```
jobs:
  build:
    name: macOS Build
    runs-on: macos-14

    steps:
      - name: Checkout repository
```
I guess you'll need to make it a bit more complicated  to ship both x86 and arm binaries for mac.

nxpfrankli, 2024-04-08:
how action know it is arm based?  

currently uuu use runs-on: macos-latest.

mwasilew, 2024-04-08:
This is a good question and I've no idea. Docs say that `runs-on: macos14` is executed on arm. From this table:
https://docs.github.com/en/actions/using-github-hosted-runners/about-github-hosted-runners/about-github-hosted-runners#supported-runners-and-hardware-resources
it seems that macos-latest also should run on arm, but it doesn't. Maybe build script should determine the architecture? It is indeed confusing.

closed 2024-04-08 by nxpfrankli