nxp-imx/mfgtools #354: uuu snap upgrade failure

dieselburner, 2023-03-17:
`sudo snap refresh` on Ubuntu 22.04.2 LTS gives following:

```
$ sudo snap refresh
error: cannot perform the following tasks:
- Run post-refresh hook of "universal-update-utility" snap if present (run hook "post-refresh": /snap/universal-update-utility/500/meta/hooks/post-refresh: 3: /snap/universal-update-utility/500/meta/hooks/post-refresh: cannot create /etc/udev/rules.d/test: Permission denied)
```

nxpfrankli, 2023-03-22:
Try again? I  remove unused hook

dieselburner, 2023-03-22:
Thanks, but you need to publish a snap package, otherwise I'm not able to try it out.

nxpfrankli, 2023-03-22:
Yes, I update it, just version have not change.

dieselburner, 2023-03-23:
Same result:
```
$ sudo snap refresh
error: cannot perform the following tasks:
- Run post-refresh hook of "universal-update-utility" snap if present (run hook "post-refresh": /snap/universal-update-utility/500/meta/hooks/post-refresh: 3: /snap/universal-update-utility/500/meta/hooks/post-refresh: cannot create /etc/udev/rules.d/test: Permission denied)

$ snap list
Name                                  Version                           Rev    Tracking         Publisher          Notes
bare                                  1.0                               5      latest/stable    canonical✓         base
...
universal-update-utility              1.4.243                           434    latest/stable    nxpfrank           -
```

dieselburner, 2023-04-01:
Had a look once again. `snap refresh` fails, while I got it through by reinstalling your package:

```
$ snap refresh
error: cannot perform the following tasks:
- Run post-refresh hook of "universal-update-utility" snap if present (run hook "post-refresh": /snap/universal-update-utility/500/meta/hooks/post-refresh: 3: /snap/universal-update-utility/500/meta/hooks/post-refresh: cannot create /etc/udev/rules.d/test: Permission denied)

$ snap remove universal-update-utility 
universal-update-utility removed

$ snap install universal-update-utility 
universal-update-utility 1.5.21 from frank li (nxpfrank) installed

$ snap refresh
All snaps up to date.
```

Closing.

closed 2023-04-01 by dieselburner