nxp-imx/mfgtools #358: uuu sometimes only flashes one block of an image.

npes-95, 2023-04-25:
Issue observed sporadically for all uuu releases > `1.4.243` on Windows 10 `10.0.19045.2846`. Log output:

```
> .\uuu.exe -v -b emmc_all <boot_binary> <image>.wic.bz2

uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.21-0-g1f42172

Build in config:
        Pctl     Chip            Vid     Pid     BcdVersion
        ==================================================
        SDPS:    MX8QXP          0x1fc9  0x012f  [0x0002..0xffff]
        SDPS:    MX8QM           0x1fc9  0x0129  [0x0002..0xffff]
        SDPS:    MX8DXL          0x1fc9  0x0147
        SDPS:    MX28            0x15a2  0x004f
        SDPS:    MX815           0x1fc9  0x013e
        SDPS:    MX865           0x1fc9  0x0146
        SDPS:    MX8ULP          0x1fc9  0x014a
        SDPS:    MX8ULP          0x1fc9  0x014b
        SDPS:    MX93            0x1fc9  0x014e
        SDP:     MX7D            0x15a2  0x0076
        SDP:     MX6Q            0x15a2  0x0054
        SDP:     MX6D            0x15a2  0x0061
        SDP:     MX6SL           0x15a2  0x0063
        SDP:     MX6SX           0x15a2  0x0071
        SDP:     MX6UL           0x15a2  0x007d
        SDP:     MX6ULL          0x15a2  0x0080
        SDP:     MX6SLL          0x1fc9  0x0128
        SDP:     MX7ULP          0x1fc9  0x0126
        SDP:     MXRT106X        0x1fc9  0x0135
        SDP:     MX8MM           0x1fc9  0x0134
        SDP:     MX8MQ           0x1fc9  0x012b
        SDPU:    SPL             0x0525  0xb4a4  [0x0000..0x04ff]
        SDPV:    SPL1            0x0525  0xb4a4  [0x0500..0x9998]
        SDPV:    SPL1            0x1fc9  0x0151  [0x0500..0x9998]
        SDPU:    SPL             0x0525  0xb4a4  [0x9999..0x9999]
        SDPU:    SPL             0x3016  0x1001  [0x0000..0x04ff]
        SDPV:    SPL1            0x3016  0x1001  [0x0500..0x9998]
        FBK:                     0x066f  0x9afe
        FBK:                     0x066f  0x9bff
        FBK:                     0x1fc9  0x0153
        FB:                      0x0525  0xa4a5
        FB:                      0x18d1  0x0d02
        FB:                      0x3016  0x0001
        FB:                      0x1fc9  0x0152
        FB:                      0x0483  0x0afb

Run built-in script:

uuu_version 1.4.149

# @_flash.bin            | bootloader, which can extract from wic image
# @_image   [_flash.bin] | wic image burn to emmc.

# This command will be run when i.MX6/7 i.MX8MM, i.MX8MQ
SDP: boot -f <boot_binary> -scanlimited 0x800000

# This command will be run when ROM support stream mode
# i.MX8QXP, i.MX8QM
SDPS: boot -scanterm -f <boot_binary> -scanlimited 0x800000

# These commands will be run when use SPL and will be skipped if no spl
# SDPU will be deprecated. please use SDPV instead of SDPU
# {
SDPU: delay 1000
SDPU: write -f <boot_binary> 0x57c00
SDPU: jump -scanlimited 0x800000
# }

# These commands will be run when use SPL and will be skipped if no spl
# if (SPL support SDPV)
# {
SDPV: delay 1000
SDPV: write -f <boot_binary> -skipspl -scanterm -scanlimited 0x800000
SDPV: jump -scanlimited 0x800000
# }

FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev ${emmc_dev}
FB: ucmd mmc dev ${emmc_dev}
FB: flash -raw2sparse all <image>.wic.bz2/*
FB: flash -scanterm -scanlimited 0x800000 <boot_binary>
FB: ucmd if env exists emmc_ack; then ; else setenv emmc_ack 0; fi;
FB: ucmd mmc partconf ${emmc_dev} ${emmc_ack} 1 0
FB: done

Wait for Known USB Device Appear...
New USB Device Attached at 1:1
1:1>Start Cmd:SDP: boot -f <boot_binary> -scanlimited 0x800000
100%1:1>Okay (1.099s)
New USB Device Attached at 1:1
1:1>Start Cmd:SDPV: delay 1000
1:1>Okay (1.008s)
1:1>Start Cmd:SDPV: write -f <boot_binary> -skipspl -scanterm -scanlimited 0x800000
100%1:1>Okay (2.865s)
1:1>Start Cmd:SDPV: jump -scanlimited 0x800000
100%1:1>Okay (1.015s)
New USB Device Attached at 1:1
1:1>Start Cmd:FB: ucmd setenv fastboot_dev mmc
1:1>Okay (0.004s)
1:1>Start Cmd:FB: ucmd setenv mmcdev ${emmc_dev}
1:1>Okay (0.005s)
1:1>Start Cmd:FB: ucmd mmc dev ${emmc_dev}
1:1>Okay (0.058s)
1:1>Start Cmd:FB: flash -raw2sparse all <image>.wic.bz2/*
16775680

```

This results in the bootloader being unable to load the kernel FIT image, and the device not booting.

nxpfrankli, 2023-04-25:
uuu extract bz2 to memory,  it takes long time some time to allocate big memory. 
Can you try 1.5.xx, which reduce memory usage for single device burn?

npes-95, 2023-04-25:
Hi, thanks for the response!

When I say uuu > 1.4.243, I mean uuu 1.5.xx, as shown in the log provided above. I've verified this behaviour for releases 1.5.4, 1.5.11 and 1.5.21. uuu 1.4.243 works as expected.

Is there anymore information I can provide to help solve the issue?

nxpfrankli, 2023-04-25:
1.5.xx have big change at memory management and multi-thread.  Can you summary some reproduce steps?  Did you met at zst file?

npes-95, 2023-04-26:
> Can you summary some reproduce steps?

The issue is not consistently reproducible (which makes sense considering the memory mgmt/threading changes you mentioned), but running `.\uuu.exe -v -b emmc_all <boot_binary> <image>.wic.bz2` a few times in a row is enough to trigger it (for me it occurs for about 2 out of 5 runs).

> Did you met at zst file?

Could you clarify what you mean here?

nxpfrankli, 2023-04-26:
New linux release switch to wic.zst format, which new compress/uncompress method,  10x fast then bz2.  Did you meet similar issue for wic.zst?