nxp-imx/mfgtools #359: linux compile err

menggaoling, 2023-05-05:
1.base64_encode need declare like this to avoid compile err.std::string HttpStream::base64_encode(std::string str);
2.
[  5%] Building CXX object libuuu/CMakeFiles/uuc_s.dir/version.cpp.o
[  8%] Linking CXX static library libuuc_s.a
[ 58%] Built target uuc_s
[ 61%] Linking CXX executable uuu
/usr/bin/ld: 找不到 -lOPENSSL_SSL_LIBRARY-NOTFOUND: 没有那个文件或目录
/usr/bin/ld: 找不到 -lOPENSSL_CRYPTO_LIBRARY-NOTFOUND: 没有那个文件或目录
make err,and no solve solution as so far.
ubuntu 64bit 22.04

nxpfrankli, 2023-05-05:
CI have not reported such error.  Did you install openssl-dev ?

menggaoling, 2023-05-06:
yes,the second problem has been solved.Thanks for your attention.

closed 2023-05-06 by menggaoling