nxp-imx/mfgtools #374: Files from the 1.5.21 release were updated with different checksum

ricardosalveti, 2023-06-02:
Any reason why the released files from the [1.5.21](https://github.com/nxp-imx/mfgtools/releases/tag/uuu_1.5.21) release were updated yesterday?

This broke some of our yocto recipes as they all have a new checksum.

nxpfrankli, 2023-06-02:
When I debug github workflow action,  accidently remove 1.5.21 release.  So I have to regenerate it.

ricardosalveti, 2023-06-05:
> When I debug github workflow action, accidently remove 1.5.21 release. So I have to regenerate it.

Got it, one other breaking change I noticed as well is that uuu_mac was also removed, mind adding it there again?

nxpfrankli, 2023-06-06:
appveyor mac build have issue recently.  wait for some days to recovery.

jluebbe, 2023-08-14:
The newer releases on https://github.com/nxp-imx/mfgtools/releases are marked as "pre-release". Is that intentional?