nxp-imx/mfgtools #375: 1489 Segmentation fault      "${@}"

Chris1452, 2023-06-05:
Hei Guys,
i am working on a raspiberry pi 3 model b+ :-
Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux(new setup)

Installed the UUU-Tool which is mentioned like here:-
https://snapcraft.io/install/universal-update-utility/raspbian

Run first time UUU :-
ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.21-0-g1f42172

uuu [-d -m -v -V] <bootloader|cmdlists|cmd>
.....

Added #  at /etc/ld.so.preload to clear the error

Installed:- 
udo apt-get install libusb-1.0-0-dev libbz2-dev libzstd-dev 

Find the device with -lsusb
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.21-0-g1f42172

Connected Known USB Devices
	Path	 Chip	 Pro	 Vid	 Pid	 BcdVersion
	==================================================
	1:112	 MX8MM	 SDP:	 0x1FC9	0x0134	 0x0101

Run:-
Snap connect universal-update-utility:removable-media

Run:-
uuu uuu.auto 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.21-0-g1f42172

Success 0    Failure 0                                                         
                                                                                
                                                                                
1:112    1/ 1 [                                      ] SDP: boot -f imx-boot.bin
/snap/universal-update-utility/501/bin/universal-update-utility-launch: line 64:  1489 Segmentation fault      "${@}"

Any advice? I can flash the image without any problems from our windows system. The Raspberry PI 4 Model B can't even find the MX8MM with lsusb, but that's a problem from raspberry i guess...

Chris1452, 2023-06-05:
hei guys, 
i looked at the ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.

Changed the /etc/ld.so.preload to:-
usr/lib/arm-linux-gnueabihf/libarmmen-**v7l**.so

Bcs the variable $PLATFROM is empty

cat /proc/cpuinfo | grep "model name"
model name	: ARMv7 Processor rev 4 (v7l)
model name	: ARMv7 Processor rev 4 (v7l)
model name	: ARMv7 Processor rev 4 (v7l)
model name	: ARMv7 Processor rev 4 (v7l)

run: 
uuu uuu.auto

ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.21-0-g1f42172

Success 0    Failure 0                                                                                                                                                                                                                       
                                                                                                                                                                                                                                              
                                                                                                                                                                                                                                              
1:113    2/ 3 [                                      ] SDPV: write -f imx-boot.bin -skipspl                                                                                                                                                   
/snap/universal-update-utility/501/bin/universal-update-utility-launch: line 64:  5596 Segmentation fault      "${@}"

Thanks in advance

Chris

nxpfrankli, 2023-06-15:
Maybe you can try our prebuild image: https://github.com/nxp-imx/mfgtools/releases/download/master/uuu_armv7

lionel, 2023-07-07:
The pre-built image link is 404.

I wound up here because I'm seeing a segmentation fault too. I'm using the current source built for armv7l:
```
Wait for Known USB Device Appear...
>Start Cmd:SDPS: boot -f /root/imx93-11x11-lpddr4x-evk/u-boot-spl.bin
New USB Device Attached at 2:1
Segmentation fault
```

nxpfrankli, 2024-11-19:
legal team currently review snap delivery.  So, I can't update snap util finish

Chris1452, 2024-11-20:
Thank you for your reply. As mentioned in the first post, I am using a Raspberry Pi 3 Model B. I am now using the current commit. However, when I try to flash the image, I get stuck.

uuu -d -v handtmann_v1.0.0_serie/uuu.auto 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.191-1-g5d77a61

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion	 Serial_No
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDPS:	 MX8ULP		 0x1fc9	 0x014a
	SDPS:	 MX8ULP		 0x1fc9	 0x014b
	SDPS:	 MX93		 0x1fc9	 0x014e
	SDPS:	 MX91		 0x1fc9	 0x0159
	SDPS:	 MX95		 0x1fc9	 0x015d
	SDPS:	 MX95		 0x1fc9	 0x015c
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPV:	 SPL1		 0x1fc9	 0x0151	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FBK:	 		 0x1fc9	 0x0153
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001
	FB:	 		 0x1fc9	 0x0152
	FB:	 		 0x0483	 0x0afb
Wait for Known USB Device Appear...
New USB Device Attached at 1:1332-
1:1332->Start Cmd:SDP: boot -f imx-boot.bin
100%1:1332->Okay (0.558s)
New USB Device Attached at 1:1332-
1:1332->Start Cmd:SDPV: delay 1000
1:1332->Okay (1.001s)
1:1332->Start Cmd:SDPV: write -f imx-boot.bin -skipspl
100%1:1332->Okay (1.415s)
1:1332->Start Cmd:SDPV: jump
100%1:1332->Okay (2.792s)
New USB Device Attached at 1:1332-150F1209DABC1514
1:1332-150F1209DABC1514>Start Cmd:FB: ucmd setenv fastboot_dev mmc
1:1332-150F1209DABC1514>Okay (0.05s)
1:1332-150F1209DABC1514>Start Cmd:FB: ucmd setenv mmcdev 2
1:1332-150F1209DABC1514>Okay (0.005s)
1:1332-150F1209DABC1514>Start Cmd:FB: ucmd mmc dev 2
1:1332-150F1209DABC1514>Okay (0.055s)
1:1332-150F1209DABC1514>Start Cmd:FB: flash -raw2sparse all handtmann.img

nxpfrankli, 2024-11-20:
At begin, uuu design for 64bit PC host only and use simple use mmap for big file.  maybe handtmann.img is too big for 32bit system.