nxp-imx/mfgtools #377: It didn't work on my m1 mac and "Fail detach kernel driver failure"

Fcromcna, 2023-06-16:
I build the source code on my m1 mac(macOS Ventura), and I get the uuu. But when I want to flash the image to my board. It didn't work. 
```
Wait for Known USB Device Appear...
New USB Device Attached at 1:11
1:11>Start Cmd:SDPS: boot -f flash.bin
1:11>Fail detach kernel driver failure(0.001s)
```

nxpfrankli, 2023-06-16:
Someone report similar issue before even at x86 platform.  Look like it works at Catalina

jameshilliard, 2023-07-14:
Maybe look at [snagboot](https://github.com/bootlin/snagboot) which seems to play a lot better with the native OSX HID drivers. It might be a good idea to add a hidapi based backend to mfgtools to better avoid conflicts with kernel drivers.

nxpfrankli, 2023-07-14:
Thanks,  hidapi can't provide hub port number, which prevent multi devices supports.

nxpfrankli, 2023-07-24:
@jameshilliard 

I tried to add HIDAPI,  only support one boards,  uuu -hidapi <your options> to tell uuu use hidapi as backend. 
I have not mac machine to do test.  Below is prebuild version:
 
https://github.com/nxp-imx/mfgtools/suites/14531068194/artifacts/822033431

nxpfrankli, 2023-07-27:
Some report, if you update libusb, it will resolve this problem. Can you try it?

mtilchen, 2023-08-09:
@nxpfrankli I built from the `hidapi` on macOS 13.1 with an M1-based MacBook Pro. `uuu` is crashing early in the script (the first command):

```
bash-3.2$ sudo uuu -hidapi script.uuu 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.109-11-g4f5ea5d

Success 0    Failure 0                                                                                                                                                      
                                                                                                                                                                             
Abort trap: 6                                                                                                                                                                
bash-3.2$ / 1 [=================76%=======>          ] SDP: boot -f u-boot.imx                                                                                               
?n??n??n 1/ 0 [                                      ]                                                                                                                       
uuu(82562,0x1e7051b40) malloc: *** error for object 0xb07ad039c000: pointer being freed was not allocated
uuu(82562,0x1e7051b40) malloc: *** set a breakpoint in malloc_error_break to debug```

mtilchen, 2023-08-09:
This is for `iMX6ULL`

nxpfrankli, 2023-08-09:
Okay, someone report uuu can work after update libusb.
 
brew info libusb
==> libusb: stable 1.0.26 (bottled), HEAD

Can you help try it?

mtilchen, 2023-08-09:
@nxpfrankli I built from the `master` branch (instead of `hidapi`) and was able to run my scripts successfully. Thanks.

nxpfrankli, 2023-08-09:
Great, thanks

mtilchen, 2023-08-10:
One thing we did notice is that `ucp` commands that are trying copy files to `t:/tmp` without an absolute path to the destination are failing:

```
sudo ../mfgtools/uuu/uuu setup.uuu
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.125-0-gaeb3490

Success 0    Failure 1

0:1     22/75 [DIRtmpcat /tmp/mmcdev`; echo 0 > /sys/] FBK: ucp  x-loader.imx t:/tmp
```

These `ucp` commands were not failing on the macOS `x86` build the last time I checked. It works if we change the above to:
```
FBK: ucp  x-loader.imx t:/tmp/x-loader.imx
```

Would you like me to open a separate issue for this?

LYC584520, 2023-08-28:
> @jameshilliard
> 
> I tried to add HIDAPI, only support one boards, uuu -hidapi to tell uuu use hidapi as backend. I have not mac machine to do test. Below is prebuild version:
> 
> https://github.com/nxp-imx/mfgtools/suites/14531068194/artifacts/822033431

It is aborted on Apple Silicon M2 machine:
`zsh: abort      sudo ./uuu example_kernel_emmc.uuu`

nxpfrankli, 2023-08-28:
Look like new libusb already this problem, can you check if libusb 1.0.26 can fix this problem instead of using libhid. 

bash-3.2$ brew info libusb
==> libusb: stable 1.0.26 (bottled), HEAD

nxpfrankli, 2023-08-28:
Can you create sperate issue for it. one ticket for one issue to track it easily.

nxpfrankli, 2023-08-29:
We can close this issue because libusb 1.0.26 can resolve this problem. If still have problem,  reopen it.

closed 2023-08-29 by nxpfrankli