nxp-imx/mfgtools #379: Random errors while flashing

aleeraser, 2023-07-13:
From a Linux host, I'm performing a flashing stress test on an iMX8 based board. I'm only flashing the bootloader (U-Boot).
I'm using `libuuu_1.5.21-0-g1f42172`.

The flashing script is dead simple and simply calls `uuu -b emmc imx-boot.bin` again and again, for 500 times.

The issue I'm seeing is that, randomly, the flashing may fail with apparently no solid reason, since the error message may change. So far I've seen the following messages:

```
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.21-0-g1f42172

Success 0    Failure 1

3:144112 1/ 1 [HID(W):LIBUSB_ERROR_IO                ] SDPS: boot -f imx-boot.bin
```

or 

```
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.21-0-g1f42172

Success 0    Failure 1

3:144112 1/ 7 [Failure claim interface               ] FB: ucmd setenv fastboot_dev mmc
```

I run tests of 500 flashing retries, and I usually have a failure rate of ~5%, i.e. ~25 failures for 500 retries.

What may be the cause of those failures?

This is somehow related to https://github.com/nxp-imx/mfgtools/issues/233.

nxpfrankli, 2023-07-13:
which imx8? which libusb version in your system?

aleeraser, 2023-07-13:
I'm using libusb v1.0.0, and I'm seeing this behavior on various boards with the following iMX8 variants: DX, DXP, QXP.

nxpfrankli, 2023-07-14:
did you run two instances of uuu?  I meet similar issue if run two instances.

aleeraser, 2023-07-16:
No, single instance but repeated 500 times, for many rounds. The (single) device which is being flashed lays untouched for the whole time.