nxp-imx/mfgtools #386: Flash both eMMC boot partitions

aleeraser, 2023-08-25:
Is it possible to flash the bootloader (U-Boot) for both boot partitions on an eMMC?

Right now, the normal `-b emmc` command only flashes the first boot partition.

nxpfrankli, 2023-08-25:
You can use customer script to do that. 

below this example: 

#FB: ucmd setenv fastboot_buffer ${loadaddr}
#FB: download -f u-boot-imx8qxp.imx
#FB: ucmd setexpr fastboot_blk ${fastboot_bytes}
#FB: ucmd setexpr fastboot_blk ${fastboot_blk} + 0x1FF
#FB: ucmd setexpr fastboot_blk ${fastboot_blk} / 0x200
#FB: ucmd mmc partconf ${emmc_dev} 1 1 1
#FB: ucmd echo ${fastboot_buffer}
#FB: ucmd echo ${fastboot_blk}
#FB: ucmd mmc write ${fastboot_buffer} 0x40  ${fastboot_blk}

aleeraser, 2023-08-25:
Thanks for the quick answer. May I ask you to elaborate further on the `fastboot_blk` var?

nxpfrankli, 2023-08-25:
Any name, just round up bytes to MMC block numbers.

aleeraser, 2023-08-25:
Understood. Is there a way to retrieve the fastboot size that is currently loaded in RAM?

aleeraser, 2023-08-28:
It seems I can't make it working.

I'm using `mmc partconf` to switch between boot0 and boot1 partitions, writing the bootcontainer ad address 0x0.
However, it looks like that even when I switch to boot1, the `mmc write` operation still writes to boot0, since as I reboot I can see that the imx-boot-ALTERNATE.bin is being run.

How can I switch the `mmc write` target to the second boot partition ?

Here is my uuu script:

```
uuu_version 1.2.39

SDPS: boot -f imx-boot.bin
SDPV: delay 1000
SDPV: write -f imx-boot.bin -skipspl
SDPV: jump

FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev $emmc_dev
FB: ucmd mmc dev $emmc_dev

FB: ucmd echo flashing first parttion

FB: ucmd setenv fastboot_buffer $loadaddr
FB: download -f imx-boot.bin
FB: ucmd setexpr fastboot_blk $fastboot_bytes
FB: ucmd setexpr fastboot_blk $fastboot_blk + 0x1FF
FB: ucmd setexpr fastboot_blk $fastboot_blk / 0x200
FB: ucmd mmc partconf $emmc_dev 1 1 1
FB: ucmd echo fastboot_buffer: $fastboot_buffer
FB: ucmd echo fastboot_blk: $fastboot_blk
FB: ucmd mmc write $fastboot_buffer 0 $fastboot_blk

FB: ucmd echo flashing second parttion

FB: download -f imx-boot-ALTERNATE.bin
FB: ucmd setexpr fastboot_blk $fastboot_bytes
FB: ucmd setexpr fastboot_blk $fastboot_blk + 0x1FF
FB: ucmd setexpr fastboot_blk $fastboot_blk / 0x200
FB: ucmd mmc partconf $emmc_dev 1 2 1
FB: ucmd mmc write $fastboot_buffer 0 $fastboot_blk

FB: Done

```

nxpfrankli, 2023-08-28:
Actually it is uboot's behavior.  Can you try add FB: ucmd mmc partconf $emmc_dev 1 1 0 before write second's partition?

aleeraser, 2023-08-28:
You mean replacing

`FB: ucmd mmc partconf $emmc_dev 1 2 1`

with 

`FB: ucmd mmc partconf $emmc_dev 1 1 0`

or simply adding it before it?

nxpfrankli, 2023-08-28:
simple adding it. 
u-boot=> help mmc
....,  suppose only one <partition_access> can be 1 

mmc partconf <dev> [[varname] | [<boot_ack> <boot_partition> <partition_access>]]
 - Show or change the bits of the PARTITION_CONFIG field of the specified device
   If showing the bits, optionally store the boot_partition field into varname

aleeraser, 2023-08-28:
Unfortunately it doesn't help.

If in U-Boot I switch partitions with `mmc partconf 0 1 1/2 0` and reboot, I can see that the correct U-Boot version starts... but I'm unable to flash it.

Those are the partitions as seen by fastboot:

```
idx 0, ptn 0 name='gpt' start=0 len=2048
idx 1, ptn 0 name='' start=0 len=0
idx 2, ptn 0 name='all' start=0 len=15384576
idx 3, ptn 0 name='bootloader' start=0 len=64512
idx 4, ptn 1 name='mmcsda1' start=8 len=1638400
idx 5, ptn 2 name='mmcsda2' start=1638408 len=1638400
idx 6, ptn 3 name='mmcsda3' start=3276808 len=9172992
```

What looks strange to me is that I can access and flash boot1 partition from linux.
Only U-Boot is unable to see it.

aleeraser, 2023-08-29:
I updated the previous script I posted and was able to flash both boot partitions. This can be closed.

closed 2023-08-29 by aleeraser

nxpfrankli, 2023-08-29:
Can you provide your work script? I can put it to FAQ wiki page.

aleeraser, 2023-08-30:
As I mentioned, I updated the script [in my previous post](https://github.com/nxp-imx/mfgtools/issues/386#issuecomment-1695640731)

nxpfrankli, 2023-08-30:
Thanks, update faq: https://github.com/nxp-imx/mfgtools/wiki/FAQ