nxp-imx/mfgtools #389: FB commands not running on IMX-28

ahsan-kaukab, 2023-09-13:
Hello, I have been working on IMX-28EVK and installed the uuu and when I connected the IMx28 controller I can see it using uuu -lsusb
[image]

When I try to run SPDS commands its running fine 
[image]

But the problem occurs when I try to run FB / FBK commands and the IMX board gives an error [Bulk(W):LIBUSB_ERROR_IO               ] FB: ucmd echo as follow:
[image]

Also the WinUSB drivers on my imxboard is as follow
[image]

The idea that I have to update the splash screen on my IMX-28 using uuu utility and uboot command but FB and FBK is not working on IMX28-EVK , any one can help me with this issue ?

I have tested it using Win10 and Ubuntu 22.0

nxpfrankli, 2023-09-13:
I am not sure how old your imx28 uboot.  Some fastboot patch need be backport

ahsan-kaukab, 2023-09-13:
Can you specify how to add patch and build bootloader for IMX28 because according to another form I can see that bootloader can be customized using CE-6 and VS-2005 which is not supported in Win10

ahsan-kaukab, 2023-09-13:
> I am not sure how old your imx28 uboot. Some fast boot patch need be backport

Can you specify how to add a patch and build uboot for IMX28 according to another forum, I can see that uboot can be customized using CE-6 and VS-2005 which is not supported in Win10 ?
Also Plz send link of patch to add in uboot

ahsan-kaukab, 2023-09-13:
> I am not sure how old your imx28 uboot. Some fastboot patch need be backport

I have source code of uboot but not able to build it on win-10 / ubuntu-20 because I can not see a good way of building uboot for imx28

ahsan-kaukab, 2023-09-13:
> It is too old. I can't remember exactly. It is similar with other uboot, you can good it. The only difference is, iMX28 need a tools to call elftosb to convert uboot to ROM supported format.

After creating uboot from
https://github.com/nxp-imx/uboot-imx

Now the elf tool is giving error as
[image]

nxpfrankli, 2023-09-13:
u-boot-imx28.bd  line 9 have problem, HAB feature is not support in imx28

nxpfrankli, 2025-04-01:
long time no update, close it

closed 2025-04-01 by nxpfrankli