nxp-imx/mfgtools #390: How to automate UUU to match both USB 2.0 and USB 3.0 port path

Dray1998, 2023-09-19:
Hi, 
When UUU uses "-m" to monitor USB path, it will get stuck during fastboot because the path has changed. 
This is mentioned in the UUU pdf file as well:
"Notes: Some board supports super speed (USB3.0). USB 3.0 port path is difference USB 2.0. If use -m to filter port, you need add USB 3.0 port number otherwise fastboot will not be detected."

I used "uuu -lsusb" to get the USB path of my device, but it does not give the USB 3.0 port number. Currently, I'm using USB device tree viewer to get both USB 2.0 and USB 3.0 port path, but its not suitable if I want to write a script to automate this process. Different PC might have different paths, so the script has to be able to determine itself the corresponding USB 3.0 port path for each detected USB 2.0 path.

May I know if you have any suggestion for this?

nxpfrankli, 2023-09-19:
using two -m option to monitor two path,  -m Path1 -m path2

Dray1998, 2023-09-19:
Hi @nxpfrankli 
Thanks for the reply. Yes, I understand that uuu can monitor multiple paths, but my question is how to determine the corresponding USB 3.0 port path for each detected USB 2.0 path by "uuu -lsusb".

nxpfrankli, 2023-09-19:
I use uuu -lsusb again after `uuu flash.bin` to determinate usb 3.0 port.  OS have not provide map info between USB 2.0 and USB 3.0 port

Dray1998, 2023-09-19:
In a scenario of running multiple devices at the same time, it it difficult to differentiate between devices using -lsusb after uuu flash.bin, so its best if can determine it before flash.bin
I'm using USB device tree viewer to get the map info between USB 2.0 and USB 3.0 port. It provides both port chain and companion port chain, which represents USB 2.0 and USB 3.0 port path. 
I noticed uuu uses the libusb library, and within it has _**libusb_get_ss_endpoint_companion_descriptor**_  function. This	function finds and parses the USB 3.0 endpoint companion descriptor given by endpoint. Maybe this could help uuu to determine the USB 3.0 port path changes during fastboot?

nxpfrankli, 2024-02-14:
try https://github.com/nxp-imx/mfgtools/commit/33f8086c3b3a5d07e1232ec90534dd82ff4ea4f5, added serial number filter support.

nxpfrankli, 2025-04-01:
long time no update, close it