nxp-imx/mfgtools #391: Fail Bulk(R):LIBUSB_ERROR_TIMEOUT(2.009s)

mhadavan, 2023-09-27:
Based on some other issues reported here #334 , 
I did a simple test to reproduce a LIBUSB_ERROR_TIMEOUT problem that I have on a board:

> SDP: boot -f imx-boot-dhunx-sd.bin-flash_evk
> SDPV: delay 1000
> SDPV: write -f imx-boot-dhunx-sd.bin-flash_evk -skipspl
> SDPV: jump
> FB[-t 15000]: ucmd sleep 3

The problem here is that I always get Fail Bulk(R):LIBUSB_ERROR_TIMEOUT during the sleep operation.

Is this a bug in uuu or is there some parameters I can change to avoid the error message?

mhadavan, 2023-09-27:
After some more investigations and testing, I have found that if I increase the m_Timeout to something big in trans.cpp:: BulkTrans:Read function, I do not get the LIB_ERROR_TIMEOUT message in my test above.

`ret = libusb_bulk_transfer(
	(libusb_device_handle *)m_devhandle,
	m_ep_in.addr,
	p,
	size,
	&actual_length,
	100000
);`

Is there any reason for having it at 2000?

nxpfrankli, 2023-09-27:
uboot command is sync.  Default 2s to run it.  BulkTrans:read try to get ucmd's running result. If exceed 2s, it will be timeout.

nxpfrankli, 2025-04-01:
long time no update, close it.

closed 2025-04-01 by nxpfrankli