nxp-imx/mfgtools #392: Issue programing multiple devices at the same time

Synergetik, 2023-09-28:
Hi,

I'm currently trying to flash multiple i.MX6 boards at the same time using release 1.5.21 on a Win10 PC. To this end I'm using multiple cmd instances calling uuu with the -m option to address specific USB devices.

Unfortunately, only one flash procedure appears to be performed at a time.
Having started one such flash procedure for one board always results in the other instances being aborted with errors "Fail HID(W):LIBUSB_ERROR_IO" or "Fail HID(W):LIBUSB_ERROR_TIMEOUT".
Using option -d allows to serialize these procedures. But what I'm aiming for is to parallelize them to save time during production.

I've already found issue #194 being very similar to what I'm facing right now and that it was solved using version 1.3.102. However, using other versions of the tool didn't help in my case.

No such issues are observed using the same setup to flash a single board.

Since one of the key features listed for the tool is "Multi devices program support" I'm quite sure that it's somehow possible to flash multiple boards at the same time. 

Any ideas on what I'm doing wrong?

nxpfrankli, 2023-09-28:
Can you post your cmd and error log? I am little confused

Synergetik, 2023-09-29:
Hi and thanks again!

'uuu -lsusb' command leads to output

```
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.21-0-g1f42172

Connected Known USB Devices
        Path     Chip    Pro     Vid     Pid     BcdVersion
        ==================================================
        6:8422   MX6ULL  SDP:    0x---- 0x----   0x0001
        6:7422   MX6ULL  SDP:    0x---- 0x----   0x0001
```

(VID and PID are blanked out)

What can be observed most of the time is the following:
For the 1st instance running the command 'uuu -d -m 6:7422 u-boot-devel.imx' leads to:

`Success 1    Failure 0     at path 6:7422`

In a seperate CMD window for the 2nd instance running the command 'uuu -d -m 6:8422 u-boot-devel.imx' immediately after having started the 1st run leads to: 

```
Success 0    Failure 1     at path 6:8422
6:8422   1/ 2 [HID(W):LIBUSB_ERROR_IO                ] SDP: boot -f "u-boot-devel.imx"
```

(LIBUSB_ERROR_IO can be observed with and without -d option)

And if this is of any help, (occasionally) even the first flash procedure fails with a timeout error:
```
Success 0    Failure 1     at path 6:7422
6:7422   1/ 2 [HID(R):LIBUSB_ERROR_TIMEOUT           ] SDP: boot -f "u-boot-devel.imx"
```

nxpfrankli, 2023-09-29:
Strange, I use this ways every days.