nxp-imx/mfgtools #393: Unexpected behavior with ucp

iandulchinos-cpi, 2023-10-06:
Using `uuu` version 1.5.21, but the issue exists on 1.4.139 as well. Haven't tried any other versions.

Copying a file and then copying again with a truncated version of the same file results in extra characters added to the resulting file on the device.

### Steps to reproduce
On the PC running `uuu`, with the device in FBK mode, run
```
$ echo abcd > /tmp/test.txt
$ uuu FBK: ucp /tmp/test.txt T:/tmp/test.txt
$ echo ab > /tmp/test.txt
$ uuu FBK: ucp /tmp/test.txt T:/tmp/test.txt
```
Expected contents of `/tmp/test.txt` on the device:
```
ab
```
Actual contents of `/tmp/test.txt` on the device:
```
ab
d
```

Is this a known issue?

nxpfrankli, 2023-10-06:
Possible it is bugs. 

Can you try
```
diff --git a/libuuu/buffer.cpp b/libuuu/buffer.cpp
index 7e32a06..0500d6e 100644
--- a/libuuu/buffer.cpp
+++ b/libuuu/buffer.cpp
@@ -239,7 +239,7 @@ public:
                if (p->mapfile(backfile.substr(1), st.st_size))
                        return -1;

-               p->m_available_size = st.st_size;
+               p->m_DataSize = p->m_available_size = st.st_size;

                atomic_fetch_or(&p->m_dataflags, FILEBUFFER_FLAG_LOADED | FILEBUFFER_FLAG_NEVER_FREE);
                p->m_request_cv.notify_all();
```

iandulchinos-cpi, 2023-10-06:
@nxpfrankli Thank you for the quick response! I added that change on top of the most recent master commit and built, still seeing the issue.