nxp-imx/mfgtools #401: Ask a question about mx6ull

shun12a, 2023-11-16:
Does mx6ull support uuu script FBK?
I made a comparison between FB and FBK and found that FB could get feedback while FBK could not. I also found on NXP's official website what NXP employees said: "seems fbk is working only with android as described in", Thank you for your advice

shun12a, 2023-11-16:
[image]

shun12a, 2023-11-16:
[image]

nxpfrankli, 2023-11-16:
Make sure your initramfs included new https://github.com/nxp-imx/imx-uuc

shun12a, 2023-11-17:
> Make sure your initramfs included new https://github.com/nxp-imx/imx-uuc
Yes, thank you for your answer. How can I replace it? I found that the official "fsl-image-mfgtool-initramfs-imx_mfgtools.cpio.gz.u-boot" does not have a C file about uuc. Do I still need to mount it?

shun12a, 2023-11-17:
[image]

shun12a, 2023-11-20:
thank for your help,I was so inspired by your answer

shun12a, 2023-11-20:
I have another question about how wic is created. I want to use their own compiled files to make wic files burned to the development board

shun12a, 2023-11-20:
[image]

shun12a, 2023-11-20:
[image]

shun12a, 2023-11-20:
> Make sure your initramfs included new https://github.com/nxp-imx/imx-uuc

cool

nxpfrankli, 2025-04-01:
Long time no update, close it

closed 2025-04-01 by nxpfrankli