nxp-imx/mfgtools #403: Fail Bulk(W):LIBUSB_ERROR_TIMEOUT

shun12a, 2023-11-24:
I have done rootfs burning, but sometimes the burning can be successful, and sometimes Fail Bulk(W):LIBUSB_ERROR_TIMEOUT error will occur

shun12a, 2023-11-24:
[image]

shun12a, 2023-11-24:
[image]

shun12a, 2023-11-24:
[image]

shun12a, 2023-11-24:
I have downloaded the latest uuc to fsl-image-mfgtool-initramfs-imx_mfgtools.cpio.gz.u-boot

shun12a, 2023-11-24:
I download by raw is no problem,What is the cause of this?

shun12a, 2023-11-24:
[image]

shun12a, 2023-11-24:
[image]

shun12a, 2023-11-24:
[image]

nxpfrankli, 2023-11-29:
It should be out of the memory.  file system takes too much memory and schedule have not time flush it to storage.