nxp-imx/mfgtools #405: "sudo uuu -v -b emmc ./imx-boot-sd.bin" stuck at 100%1:12>Okay (0.646s) on MX8MM

anfigueroa, 2023-12-05:
When running

```sudo uuu -v -b emmc ./imx-boot-sd.bin```

- Using uuu version 1.5.125
- Executing on Debian 12.2
- Running on Raspberry Pi 4

We get

```
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.141-3-g3a5beb6

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDPS:	 MX8ULP		 0x1fc9	 0x014a
	SDPS:	 MX8ULP		 0x1fc9	 0x014b
	SDPS:	 MX93		 0x1fc9	 0x014e
	SDPS:	 MX95		 0x1fc9	 0x015d
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPV:	 SPL1		 0x1fc9	 0x0151	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FBK:	 		 0x1fc9	 0x0153
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001
	FB:	 		 0x1fc9	 0x0152
	FB:	 		 0x0483	 0x0afb

Run built-in script:
 
uuu_version 1.2.39

# @_flash.bin            | bootloader
# @_image   [_flash.bin] | image burn to emmc, default is the same as bootloader

# This command will be run when i.MX6/7 i.MX8MM, i.MX8MQ
SDP: boot -f imx-boot-sd.bin

# This command will be run when ROM support stream mode
# i.MX8QXP, i.MX8QM
SDPS: boot -f imx-boot-sd.bin

# These commands will be run when use SPL and will be skipped if no spl
# SDPU will be deprecated. please use SDPV instead of SDPU
# {
SDPU: delay 1000
SDPU: write -f imx-boot-sd.bin -offset 0x57c00
SDPU: jump
# }

# These commands will be run when use SPL and will be skipped if no spl
# if (SPL support SDPV)
# {
SDPV: delay 1000
SDPV: write -f imx-boot-sd.bin -skipspl
SDPV: jump
# }

FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev ${emmc_dev}
FB: ucmd mmc dev ${emmc_dev}
FB: flash bootloader imx-boot-sd.bin
FB: ucmd if env exists emmc_ack; then ; else setenv emmc_ack 0; fi;
FB: ucmd mmc partconf ${emmc_dev} ${emmc_ack} 1 0
FB: Done

Wait for Known USB Device Appear...
New USB Device Attached at 1:12
1:12>Start Cmd:SDP: boot -f imx-boot-sd.bin
100%1:12>Okay (0.646s)
```
Sometimes we get
```
>Start Cmd:SDP: boot -f imx-boot-sd.bin
Wait for Known USB /
Wait for Known USB |
```

**This same execution works perfectly fine in a M1 Macbook and in a Raspberry Pi Zero with same OS and versions.**

We used the same SD card from the Pi Zero to boot on the Pi 4 and it still wouldn't work. This leads us to think that there might be something that uuu is not able to do properly on the Pi 4. We also tried different Pi 4 boards, none work yet.

Any help would be much appreciated.

agusgonzaleznic, 2023-12-06:
Hey! im having the same issue! It's just not working upon write on a raspberry pi 4b

anfigueroa, 2023-12-06:
Adding execution with libusb debug:

```
pi@flashing-tool-02:~/mfgtools/uuu $ ./uuu -v -V -b emmc imx-boot-imx8mmevk-sd.bin-flash_evk
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000316] [0000061d] libusb: debug [libusb_init] created default context
[ 0.000751] [0000061d] libusb: debug [libusb_init] libusb v1.0.26.11724
[ 0.000918] [0000061d] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.001050] [0000061d] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.001141] [0000061d] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.001259] [0000061d] libusb: debug [get_kernel_version] reported kernel version is 6.1.0-rpi6-rpi-v8
[ 0.001634] [0000061d] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.001680] [0000061d] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.001738] [0000061d] libusb: debug [op_init] sysfs is available
[ 0.002833] [0000061e] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.022079] [0000061d] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.022145] [0000061d] libusb: debug [linux_get_device_address] scan usb1
[ 0.022489] [0000061d] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.022595] [0000061d] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.022689] [0000061d] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.023708] [0000061d] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.023816] [0000061d] libusb: debug [linux_get_device_address] scan 1-1
[ 0.024123] [0000061d] libusb: debug [linux_get_device_address] bus=1 dev=2
[ 0.024223] [0000061d] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[ 0.024315] [0000061d] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[ 0.024672] [0000061d] libusb: debug [linux_get_parent_info] dev 0x5599f7bce0 (1-1) has parent 0x5599f9f200 (usb1) port 1
[ 0.025659] [0000061d] libusb: debug [linux_get_device_address] getting address for device: 1-1.2 detached: 0
[ 0.025774] [0000061d] libusb: debug [linux_get_device_address] scan 1-1.2
[ 0.026074] [0000061d] libusb: debug [linux_get_device_address] bus=1 dev=3
[ 0.026174] [0000061d] libusb: debug [linux_enumerate_device] busnum 1 devaddr 3 session_id 259
[ 0.026264] [0000061d] libusb: debug [linux_enumerate_device] allocating new device for 1/3 (session 259)
[ 0.026558] [0000061d] libusb: debug [linux_get_parent_info] dev 0x5599f9f440 (1-1.2) has parent 0x5599f7bce0 (1-1) port 2
[ 0.027400] [0000061d] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.027509] [0000061d] libusb: debug [linux_get_device_address] scan usb2
[ 0.027799] [0000061d] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.027897] [0000061d] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.027987] [0000061d] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.141-3-g3a5beb6

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDPS:	 MX8ULP		 0x1fc9	 0x014a
	SDPS:	 MX8ULP		 0x1fc9	 0x014b
	SDPS:	 MX93		 0x1fc9	 0x014e
	SDPS:	 MX95		 0x1fc9	 0x015d
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPV:	 SPL1		 0x1fc9	 0x0151	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
[... 128 lines cut ...]
[11.042480] [0000061f] libusb: debug [usbi_wait_for_events] poll() returned 1
[11.042537] [0000061f] libusb: debug [reap_for_handle] urb type=1 status=-2 transferred=0
[11.042600] [0000061f] libusb: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
[11.042639] [0000061f] libusb: debug [handle_bulk_completion] abnormal reap: urb status -2
[11.042672] [0000061f] libusb: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting

[5:05](https://jucrworkspace.slack.com/archives/C04J95S90BD/p1701806711440889)
[11.042705] [0000061f] libusb: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
[11.042763] [0000061f] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[11.042808] [0000061f] libusb: debug [usbi_handle_transfer_completion] transfer 0x7fa0001d78 has callback 0x7fa798cd40
[11.042845] [0000061f] libusb: debug [sync_transfer_cb] actual_length=0
[11.042919] [0000061f] libusb: debug [libusb_free_transfer] transfer 0x7fa0001d78
1:12>Fail HID(W):LIBUSB_ERROR_TIMEOUT(10.01s)
[11.043899] [0000061f] libusb: debug [libusb_close]  
[... cut ...]
[11.147792] [0000061e] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
[11.148089] [0000061d] libusb: debug [libusb_unref_device] destroy device 2.1
[11.148140] [0000061d] libusb: debug [libusb_unref_device] destroy device 1.3
[11.148176] [0000061d] libusb: debug [libusb_unref_device] destroy device 1.2
[11.148213] [0000061d] libusb: debug [libusb_unref_device] destroy device 1.1
[11.148249] [0000061d] libusb: debug [usbi_remove_event_source] remove fd 4
[11.148315] [0000061d] libusb: debug [usbi_remove_event_source] remove fd 3
```

capiman, 2024-04-29:
Perhaps worth trying out if latest version works for you...

Fix was done here:
https://github.com/nxp-imx/mfgtools/issues/420