nxp-imx/mfgtools #410: LIBUSB_ERROR_TIMEOUT and retry strategy

jmlemetayer, 2024-01-29:
Hi,

I am using a `FBK: ucp <filename>` command with a 100MB+ file.

This is leading to a well known `Fail Bulk(W):LIBUSB_ERROR_TIMEOUT` most of the time (> 95% failure).

I try to increase the timeout but that didn't help. I found that doing a small retry in the `FastBoot::Transport` fix my issue:
```diff
--- a/libuuu/fastboot.cpp
+++ b/libuuu/fastboot.cpp
@@ -65,7 +65,8 @@ int FastBoot::Transport(string cmd, void *p, size_t size, vector<uint8_t> *input
                size_t actual;
                memset(buff, 0, 65);
                if (m_pTrans->read(buff, 64, &actual))
-                       return -1;
+                       if (m_pTrans->read(buff, 64, &actual))
+                               return -1;
 
                if (strncmp(buff, "DATA",4) == 0)
                {
```

I have found a lot of issue mentionning the `LIBUSB_ERROR_TIMEOUT` either with `Bulk` or `HID` and I think a retry strategy can solve them all.

The retry strategy can be more specific or more generic, but I have trouble deciding between:

1. Apply my previous patch as-is (very specific)
2. Update `BulkTrans::read` to add a retry
3. Add a  global retry strategy to the `TransBase` class
4. Add a global retry strategy on both read and write (very generic)

I would go for the 4th solution. Do you have any ideas about the best solution?

I would be able to propose a patch later.

nxpfrankli, 2024-01-29:
Actually, I hope can found the root cause why TIMEOUT happen.  I suppose client side have not feed data in time or libusb have bugs.

jmlemetayer, 2024-01-29:
During the `FBK` operation i am using the https://github.com/nxp-imx/imx-uuc daemon. So I have added some debug traces, but I didn't see nothing.

The last bulk was ok, and then nothing ... Seems more like a host issue, and maybe very low level (libusb or kernel).

For now I have pushed the new experimental `uuu` to my CI runner. It seems to work well. Let's wait some time to see if it's stable.

jmlemetayer, 2024-02-09:
I've been using the proposed fix for 2 weeks on my CI runner and manually. I didn't see any more USB timeouts or other errors.

I think this fix can be safely merged.

PR #411 (2024-02-12): libuuu: implement a basic retry strategy for usb transfer
This commit fixes #410 and maybe others like #403, #341, #339, ...
closed 2024-02-12 by nxpfrankli