nxp-imx/mfgtools #412: UBUNTU 20.04.6 LTS; uuu: Fail HID(W):LIBUSB_ERROR_NO_DEVICE(0.003s)

brenkem, 2024-01-30:
Hello,

I am unable to perform a uboot recovery on a IMX8MP chip in Ubuntu 20.04 (see [0]). But when I start the recovery process under Windows the idVendor changed  from [1] to [2] and I am able to run uuu successful under Linux. I noticed that the dmesg output changed from "Product: SE Blank 865", when I manually enter the "USB Serial Download Mode" to "Product: USB download gadget" after running the (uuu.auto script [5]) under Windows.

How do I set up the chip to be noticed from LINUX directly as [2]. Or is there any other way to fix [0]?

[0]:
`$ sudo uuu -v .
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.163-0-g534bf1f

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	…
	FB:	 		 0x0483	 0x0afb
Wait for Known USB Device Appear...
New USB Device Attached at 3:4
3:4>Start Cmd:SDPS: boot -f nboot.fs
3:4>**Fail HID(W):LIBUSB_ERROR_NO_DEVICE(0.003s)**
`

[1]:
`$ sudo uuu -lsusb
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.163-0-g534bf1f

Connected Known USB Devices
	Path	 Chip	 Pro	 Vid	 Pid	 BcdVersion
	==================================================
	3:4	 MX865	 SDPS:	 0x1FC9	0x0146	 0x0002
`

[2]:
`$ sudo uuu -lsusb
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.163-0-g534bf1f

Connected Known USB Devices
	Path	 Chip	 Pro	 Vid	 Pid	 BcdVersion
	==================================================
	3:4	 	 FB:	 0x0525	0xA4A5	 0x0223
`

[3]:
`$ dmesg

[ 7222.693005] usb 3-4: USB disconnect, device number 25
[ 7224.793742] usb 3-4: new high-speed USB device number 26 using xhci_hcd
[ 7224.943449] usb 3-4: New USB device found, idVendor=1fc9, idProduct=0146, bcdDevice= 0.02
[ 7224.943461] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 7224.943465] usb 3-4: Product: SE Blank 865  
[ 7224.943468] usb 3-4: Manufacturer: NXP       SemiConductor Inc 
[ 7224.946570] hid-generic 0003:1FC9:0146.0017: hiddev0,hidraw1: USB HID v1.10 Device [NXP       SemiConductor Inc  SE Blank 865  ] on usb-0000:00:14.0-4/input0
`

[4]:
`$ dmesg

[ 7186.926056] usb 3-4: new high-speed USB device number 25 using xhci_hcd
[ 7187.076066] usb 3-4: New USB device found, idVendor=0525, idProduct=a4a5, bcdDevice= 2.23
[ 7187.076078] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7187.076082] usb 3-4: Product: USB download gadget
[ 7187.076085] usb 3-4: Manufacturer: FSL
[ 7187.076087] usb 3-4: SerialNumber: 1a05180089f509ed
[ 7187.078401] usb-storage 3-4:1.0: USB Mass Storage device detected
[ 7187.079050] usb-storage 3-4:1.0: Quirks match for vid 0525 pid a4a5: 10000
[ 7187.127797] usb-storage 3-4:1.0: USB Mass Storage device detected
[ 7187.127937] usb-storage 3-4:1.0: Quirks match for vid 0525 pid a4a5: 10000
`

[5]:
[uuu.auto.txt](https://github.com/nxp-imx/mfgtools/files/14102199/uuu.auto.txt)

nxpfrankli, 2024-01-30:
Your uboot is old.  we already changed FB PID/VID, 0525 and a4a5 is mass storage device, which used by linux default gadget.

brenkem, 2024-02-08:
Hi. I discussed the issue with the manufacturer and they told me that they had a custom uuu version to solve the issue.

The issue is therefore solved.

Thanks for the support.

closed 2024-02-08 by brenkem