nxp-imx/mfgtools #423: iMX6UL Fail Bulk(W): LIBUSB_ERROR_TIMEOUT (-7) with big rootfs

swechencheng, 2024-05-03:
Hi, I am trying uuu (1.5.181) on a custom board based on iMX6UL and flash rootfs on NAND.
I build both core-image-minimal and core-image-full-cmdline with Yocto.
I am based on the example nand script:
```
uuu_version 1.4.193

# This command will be run when i.MX6/7 i.MX8MM, i.MX8MQ
SDP: boot -f skyline410/u-boot-skyline410.imx-nand

# These commands will be run when use SPL and will be skipped if no spl
# if (SPL support SDPV)
# {
SDPV: delay 1000
SDPV: write -f skyline410/u-boot-skyline410.imx-nand -skipspl
SDPV: jump
# }

FB: ucmd printenv

FB: ucmd setenv fastboot_buffer ${loadaddr}
FB: download -f skyline410/zImage-skyline410.bin
FB: ucmd setenv fastboot_buffer ${fdt_addr}
FB: download -f skyline410/imx6ul-14x14-evk-gpmi-weim-skyline410.dtb
FB: ucmd setenv fastboot_buffer ${initrd_addr}
FB: download -f skyline410/skyline-mfgtool-initramfs-skyline410.cpio.zst.u-boot
FB: ucmd setenv bootargs ${bootargs} ${mtdparts}
#FB: ucmd setenv bootargs console=ttymxc3,115200  ${mtdparts}
FB: acmd ${kboot} ${loadaddr} ${initrd_addr} ${fdt_addr}

FBK: ucmd cat /proc/mtd
FBK: ucmd cat /proc/mtd | while read dev size erase name; do mtd=${dev:3}; mtd=${mtd%:}; name=${name%\"}; name=${name#\"}; echo export $name=$mtd >> /tmp/mtd.sh; done;

FBK: ucmd chmod 777 /tmp/mtd.sh
FBK: ucmd mount -t debugfs debugfs /sys/kernel/debug

# burn rootfs
FBK: ucmd source /tmp/mtd.sh; flash_erase /dev/mtd${nandrootfs} 0 0
FBK: ucmd source /tmp/mtd.sh; ubiformat /dev/mtd${nandrootfs}
FBK: ucmd source /tmp/mtd.sh; ubiattach /dev/ubi_ctrl -m ${nandrootfs}
FBK: ucmd source /tmp/mtd.sh; ubimkvol /dev/ubi0 -Nnandrootfs -m
FBK: ucmd source /tmp/mtd.sh; mkdir -p /mnt/mtd
FBK: ucmd source /tmp/mtd.sh; mount -t ubifs ubi0:nandrootfs /mnt/mtd
FBK: acmd export EXTRACT_UNSAFE_SYMLINKS=1; tar x -z -C /mnt/mtd
#FBK: ucp skyline410/core-image-minimal-skyline410.tar.gz t:-
FBK: ucp skyline410/core-image-full-cmdline-skyline410.tar.gz t:-
#FBK: ucp skyline410/test-benchmark-skyline410.tar.gz t:-
FBK: sync
FBK: ucmd umount /mnt/mtd

FBK: done
``` 
Flashing core-image-minimal is fine, however, `FBK: ucmd umount /mnt/mtd` is very slow.

But when flashing core-image-full-cmdline, it fails:
```
~/mfgtool/binaries/kirkstone $ sudo uuu -s nand.rootfs.skyline410.uuu
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.181-0-gd362f4a

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion	 Serial_No
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDPS:	 MX8ULP		 0x1fc9	 0x014a
	SDPS:	 MX8ULP		 0x1fc9	 0x014b
	SDPS:	 MX93		 0x1fc9	 0x014e
[... 415 lines cut ...]
[   17.402305] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
[   17.409216] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
[   17.416703] ubi0: good PEBs: 2560, bad PEBs: 0, corrupted PEBs: 0
[   17.423192] ubi0: user volume: 0, internal volumes: 1, max. volumes count: 128
[   17.430602] ubi0: max/mean erase counter: 0/0, WL threshold: 4096, image sequence number: 713225854
[   17.440071] ubi0: available PEBs: 2474, total reserved PEBs: 86, PEBs reserved for bad PEB handling: 80
[   17.449717] ubi0: background thread "ubi_bgt0d" started, PID 224
run shell cmd: source /tmp/mtd.sh; ubimkvol /dev/ubi0 -Nnandrootfs -m
run shell cmd: source /tmp/mtd.sh; mkdir -p /mnt/mtd
run shell cmd: source /tmp/mtd.sh; mount -t ubifs ubi0:nandrootfs /mnt/mtd
[   17.585897] UBIFS (ubi0:0): default file-system created
[   17.592219] UBIFS (ubi0:0): Mounting in unauthenticated mode
[   17.603871] UBIFS (ubi0:0): background thread "ubifs_bgt0_0" started, PID 228
[   17.646041] UBIFS (ubi0:0): UBIFS: mounted UBI device 0, volume 0, name "nandrootfs"
[   17.654242] UBIFS (ubi0:0): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
[   17.664615] UBIFS (ubi0:0): FS size: 312487936 bytes (298 MiB, 2461 LEBs), max 2474 LEBs, journal size 15618048 bytes (14 MiB, 123 LEBs)
[   17.677334] UBIFS (ubi0:0): reserved for root: 4952683 bytes (4836 KiB)
[   17.684341] UBIFS (ubi0:0): media format: w5/r0 (latest is w5/r0), UUID 628F74BD-9F15-441F-BDB5-3B5BBB15CDB0, small LPT model
run shell cmd: export EXTRACT_UNSAFE_SYMLINKS=1; tar x -z -C /mnt/mtd
WOpen:-
[   64.498032] cfg80211: failed to load regulatory.db
```

This is a very steady failure which fails exactly at the same 55%.
Thanks in advance if anyone could help to look at this.

nxpfrankli, 2024-05-04:
we recent meet similar problem at small me
mory devices. 

try patch linux kernel driver

Changed the page alignment check within the prepare_td_for_sg() function
since the first buffer could not be page size aligned, so the check is
only performed for the following buffers.

---
 drivers/usb/chipidea/udc.c | 3 ++-
 1 file changed, 2 insertions(+), 1 deletion(-)

diff --git a/drivers/usb/chipidea/udc.c b/drivers/usb/chipidea/udc.c
index 0b7bd3c643c3..13cbf2e2aeae 100644
--- a/drivers/usb/chipidea/udc.c
+++ b/drivers/usb/chipidea/udc.c
@@ -517,7 +517,8 @@ static int prepare_td_for_sg(struct ci_hw_ep *hwep, struct ci_hw_req *hwreq)
 	}
 
 	while (i++ < req->num_mapped_sgs) {
-		if (sg_dma_address(s) % PAGE_SIZE) {
+		/* The first buffer could be not page size aligned. */
+		if ((sg_dma_address(s) % PAGE_SIZE) && (i > 1)) {
 			dev_err(hwep->ci->dev, "not page aligned sg buffer\n");
 			return -EINVAL;
 		}

swechencheng, 2024-05-06:
@nxpfrankli Thanks for your quick reply.
I patched linux-imx-mfgtool which is used by initramfs.
However, I got same uuu failure and on the target I got:
```
free(): invalid pointer
[  575.424621] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000004
[  575.433966] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000004 ]---
```

nxpfrankli, 2024-05-06:
Can you put all kernel log, which show what kernel panic?

swechencheng, 2024-05-07:
On initramfs I completely turn off swap by `CONFIG_SWAP=n`, but no luck and still `Fail Bulk(W): LIBUSB_ERROR_TIMEOUT (-7)`.

nxpfrankli, 2024-05-09:
can you try, chipidea driver have issues.  https://github.com/nxp-imx/imx-uuc/commit/d4c6ce886c4c392bc2ab70efe5229d6ea805cf1e

swechencheng, 2024-05-10:
@nxpfrankli Thanks for the suggestion. Unfortunately, I tried adding both [uuc patch](https://github.com/nxp-imx/imx-uuc/commit/d4c6ce886c4c392bc2ab70efe5229d6ea805cf1e)
and kernel patch:
```
Changed the page alignment check within the prepare_td_for_sg() function
since the first buffer could not be page size aligned, so the check is
only performed for the following buffers.

---
 drivers/usb/chipidea/udc.c | 3 ++-
 1 file changed, 2 insertions(+), 1 deletion(-)

diff --git a/drivers/usb/chipidea/udc.c b/drivers/usb/chipidea/udc.c
index 0b7bd3c643c3..13cbf2e2aeae 100644
--- a/drivers/usb/chipidea/udc.c
+++ b/drivers/usb/chipidea/udc.c
@@ -517,7 +517,8 @@ static int prepare_td_for_sg(struct ci_hw_ep *hwep, struct ci_hw_req *hwreq)
 	}
 
 	while (i++ < req->num_mapped_sgs) {
-		if (sg_dma_address(s) % PAGE_SIZE) {
+		/* The first buffer could be not page size aligned. */
+		if ((sg_dma_address(s) % PAGE_SIZE) && (i > 1)) {
 			dev_err(hwep->ci->dev, "not page aligned sg buffer\n");
 			return -EINVAL;
 		}
```
to initramfs, but still have the same problem.

swechencheng, 2024-05-10:
Just FYI I did an experiment by changing the tar extraction to a temp fs first instead of directly writing to mtd partition:
```
adminrd@rpi-hms-nkg001:~/mfgtool/binaries/kirkstone $ cat nand.rootfs.skyline410.uuu
uuu_version 1.3.191

# This command will be run when i.MX6/7 i.MX8MM, i.MX8MQ
SDP: boot -f skyline410/u-boot-skyline410.imx-nand

# These commands will be run when use SPL and will be skipped if no spl
# if (SPL support SDPV)
# {
SDPV: delay 1000
SDPV: write -f skyline410/u-boot-skyline410.imx-nand -skipspl
SDPV: jump
# }

FB: ucmd printenv

FB: ucmd setenv fastboot_buffer ${loadaddr}
FB: download -f skyline410/zImage-skyline410.bin
FB: ucmd setenv fastboot_buffer ${fdt_addr}
FB: download -f skyline410/imx6ul-14x14-evk-gpmi-weim-skyline410.dtb
FB: ucmd setenv fastboot_buffer ${initrd_addr}
FB: download -f skyline410/skyline-mfgtool-initramfs-skyline410.cpio.zst.u-boot
FB: ucmd setenv bootargs ${bootargs} ${mtdparts}
#FB: ucmd setenv bootargs console=ttymxc3,115200  ${mtdparts}
FB: acmd ${kboot} ${loadaddr} ${initrd_addr} ${fdt_addr}

FBK: ucmd cat /proc/mtd
FBK: ucmd cat /proc/mtd | while read dev size erase name; do mtd=${dev:3}; mtd=${mtd%:}; name=${name%\"}; name=${name#\"}; echo export $name=$mtd >> /tmp/mtd.sh; done;

FBK: ucmd chmod 777 /tmp/mtd.sh
#FBK: ucmd mount -t debugfs debugfs /sys/kernel/debug

# burn rootfs
#FBK: ucmd rm -rf /tmp/*
FBK: acmd export EXTRACT_UNSAFE_SYMLINKS=1; tar x -z
FBK: ucp skyline410/test-benchmark-skyline410.tar.gz t:/tmp/rootfs
FBK: sync

FBK: ucmd source /tmp/mtd.sh; flash_erase /dev/mtd${nandrootfs} 0 0
FBK: ucmd source /tmp/mtd.sh; ubiformat /dev/mtd${nandrootfs}
FBK: ucmd source /tmp/mtd.sh; ubiattach /dev/ubi_ctrl -m ${nandrootfs}
FBK: ucmd source /tmp/mtd.sh; ubimkvol /dev/ubi0 -Nnandrootfs -m
FBK: ucmd source /tmp/mtd.sh; mkdir -p /mnt/mtd
FBK: ucmd source /tmp/mtd.sh; mount -t ubifs ubi0:nandrootfs /mnt/mtd
FBK: acmd export EXTRACT_UNSAFE_SYMLINKS=1; tar x -z -C /mnt/mtd

FBK: ucmd source /tmp/mtd.sh; nandwrite -p /dev/mtd${nandrootfs} -p /tmp/rootfs
FBK: sync
FBK: ucmd umount /mnt/mtd

FBK: done
```
The uuu log:
```
adminrd@rpi-hms-nkg001:~/mfgtool/binaries/kirkstone $ sudo uuu -s nand.rootfs.skyline410.uuu
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.181-0-gd362f4a

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion	 Serial_No
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDPS:	 MX8ULP		 0x1fc9	 0x014a
	SDPS:	 MX8ULP		 0x1fc9	 0x014b
[... 364 lines cut ...]
[    3.635814] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    3.663467] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    3.670323] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    3.679357] clk: Not disabling unused clocks
[    3.686650] Freeing unused kernel image (initmem) memory: 1024K
[    3.692733] platform regulatory.0: Falling back to sysfs fallback for: regulatory.db
[    3.722164] Run /linuxrc as init process
Found New UDC: ci_hdrc.0
ci_hdrc.0 0
ffs.utp0
[    3.905992] file system registered
run utp at /dev/usb-utp0/ep0
.
uuu fastboot client 1.0.0 [built [    3.969001] read descriptors
May  9 2024 18:50:12]
Start init[    3.975842] read strings
 usb
write string
Start handle command
run shell cmd: cat /proc/mtd
run shell cmd: cat /proc/mtd | while read dev size erase name; do mtd=${dev:3}; mtd=${mtd%:}; name=${name%\"}; name=${name#\"}; echo export $name=$mtd >> /tmp/mtd.sh; done;
run shell cmd: chmod 777 /tmp/mtd.sh
run shell cmd: export EXTRACT_UNSAFE_SYMLINKS=1; tar x -z
WOpen:/tmp/rootfs
```

swechencheng, 2024-08-09:
Hi again, I have migrated our Yocto build system to Scarthgap (u-boot-imx 2023.04; linux-imx 6.6)and now this problem is gone.

closed 2024-08-09 by swechencheng