nxp-imx/mfgtools #424: Boot failure when flashing compressed .wic image

landonh12, 2024-05-08:
Hi,

I have a report of a failure to boot to Linux after flashing 8ULP with a .wic.zst image.
The board successfully boots when a decompressed .wic image is used.

Boot log after flashing .wic.zst:
`U-Boot SPL 2023.04+gf8a2983ec8+p0 (Mar 04 2024 - 07:25:04 +0000)
Normal Boot
ELE firmware version 1.0.0-344acb84
upower_apd_inst_isr: entry
upower_init: soc_id=48
upower_init: version:11.11.13
upower_init: start uPower RAM service
user_upwr_rdy_callb: soc=b
user_upwr_rdy_callb: RAM version:12.18
Turning on switches...
Turn on switches ok
Turning on memories...
Turn on memories ok
Clearing DDR retention...
Clear DDR retention ok
SEC0:  RNG instantiated
Trying to boot from BOOTROM
Boot Stage: Primary boot
image offset 0x0, pagesize 0x200, ivt offset 0x0
Load image from 0x45000 by ROM_API
NOTICE:  BL31: v2.8(release):lf-6.6.3-1.0.0-0-g8dbe28631
NOTICE:  BL31: Built : 17:57:56, Jan 22 2024
NOTICE:  upower_init: start uPower RAM service
NOTICE:  user_upwr_rdy_callb: soc=b
NOTICE:  user_upwr_rdy_callb: RAM version:12.18

U-Boot 2023.04+gf8a2983ec8+p0 (Mar 04 2024 - 07:25:04 +0000)

M33 Sync: OK
CPU:   i.MX8ULP(Dual 7) rev1.2 at 800MHz
CPU current temperature: 31
Reset cause: POR
Boot mode: Single boot
Model: NXP i.MX8ULP EVK
DRAM:  2 GiB
Core:  54 devices, 28 uclasses, devicetree: separate
MMC:   FSL_SDHC: 0
Loading Environment from MMC... *** Warning - bad CRC, using default environment

[*]-Video Link 0 (720 x 1280)
	[0] display-controller@2e050000, video
	[1] dsi@2db00000, video_bridge
	[2] panel@0, panel
In:    serial
Out:   serial
Err:   serial
SEC0:  RNG instantiated
switch to partitions #0, OK
mmc0(part 0) is current device
UID 0x767b40ab,0xc94787d6,0x2c0535b6,0xca1dbf48
flash target is MMC:0
Net:   eth0: ethernet@29950000
Fastboot: Normal
Normal Boot
Hit any key to stop autoboot:  2  1  0 
starting USB...
Bus usb@29900000: Port not available.
Bus usb@29920000: Port not available.
USB is stopped. Please issue 'usb start' first.
switch to partitions #0, OK
mmc0(part 0) is current device
** No partition table - mmc 0 **
Couldn't find partition mmc 0:1
Running BSP bootcmd ...
switch to partitions #0, OK
mmc0(part 0) is current device
** No partition table - mmc 0 **
Couldn't find partition mmc 0:1
Can't set block device
** No partition table - mmc 0 **
Couldn't find partition mmc 0:1
Can't set block device
Booting from net ...
ethernet@29950000 Waiting for PHY auto negotiation to complete......................................... TIMEOUT !
Could not initialize PHY ethernet@29950000
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4
BOOTP broadcast 5
BOOTP broadcast 6
BOOTP broadcast 7
BOOTP broadcast 8
BOOTP broadcast 9
BOOTP broadcast 10
BOOTP broadcast 11
BOOTP broadcast 12
BOOTP broadcast 13
BOOTP broadcast 14
BOOTP broadcast 15
BOOTP broadcast 16
BOOTP broadcast 17

Retry time exceeded; starting again
ethernet@29950000 Waiting for PHY auto negotiation to complete......................................... TIMEOUT !
Could not initialize PHY ethernet@29950000
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4
BOOTP broadcast 5
BOOTP broadcast 6
BOOTP broadcast 7
BOOTP broadcast 8
BOOTP broadcast 9
BOOTP broadcast 10
BOOTP broadcast 11
BOOTP broadcast 12
BOOTP broadcast 13
BOOTP broadcast 14
BOOTP broadcast 15
BOOTP broadcast 16
BOOTP broadcast 17

Retry time exceeded; starting again
WARN: Cannot load the DT
=> ls
ls - list files in a directory (default /)

Usage:
ls <interface> [<dev[:part]> [directory]]
    - List files in directory 'directory' of partition 'part' on
      device type 'interface' instance 'dev'.
=> mmc info
Device: FSL_SDHC
Manufacturer ID: 45
OEM: 0
Name: DA6032 
Bus Speed: 194641920
Mode: HS400ES (200MHz)
Rd Block Len: 512
MMC version 5.1
High Capacity: Yes
Capacity: 29.1 GiB
Bus Width: 8-bit DDR
Erase Group Size: 512 KiB
HC WP Group Size: 8 MiB
User Capacity: 29.1 GiB WRREL
Boot Capacity: 4 MiB ENH
RPMB Capacity: 4 MiB ENH
Boot area 0 is not write protected
Boot area 1 is not write protected
=> `

Boot log after flashing .wic:
`U-Boot 2023.04+gf8a2983ec8+p0 (Mar 04 2024 - 07:25:04 +0000)

M33 Sync: OK
CPU:   i.MX8ULP(Dual 7) rev1.2 at 800MHz
CPU current temperature: 32
Reset cause: POR
Boot mode: Single boot
Model: NXP i.MX8ULP EVK
DRAM:  2 GiB
Core:  54 devices, 28 uclasses, devicetree: separate
MMC:   FSL_SDHC: 0
Loading Environment from MMC... *** Warning - bad CRC, using default environment

[*]-Video Link 0 (720 x 1280)
	[0] display-controller@2e050000, video
	[1] dsi@2db00000, video_bridge
	[2] panel@0, panel
In:    serial
Out:   serial
Err:   serial
SEC0:  RNG instantiated
switch to partitions #0, OK
mmc0(part 0) is current device
UID 0x767b40ab,0xc94787d6,0x2c0535b6,0xca1dbf48
flash target is MMC:0
Net:   eth0: ethernet@29950000
Fastboot: Normal
Normal Boot
Hit any key to stop autoboot:  2  1  0 
starting USB...
Bus usb@29900000: Port not available.
Bus usb@29920000: Port not available.
USB is stopped. Please issue 'usb start' first.
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1...
36322 bytes read in 3 ms (11.5 MiB/s)
Working FDT set to 83000000
7[r[999;999H[6n8No EFI system partition
No EFI system partition
Failed to persist EFI variables
BootOrder not defined
EFI boot manager: Cannot load any image
Running BSP bootcmd ...
switch to partitions #0, OK
mmc0(part 0) is current device
Failed to load 'boot.scr'
34877952 bytes read in 132 ms (252 MiB/s)
Booting from mmc ...
36322 bytes read in 8 ms (4.3 MiB/s)
## Flattened Device Tree blob at 83000000
   Booting using the fdt blob at 0x83000000
Working FDT set to 83000000
   Using Device Tree in place at 0000000083000000, end 000000008300bde1
Working FDT set to 83000000

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd040]`

allenxh, 2024-05-08:
> [1] ***@***.***, video_bridge
> Bus ***@***.***: Port not available.
> ls [<dev[:part]> [directory]]

Did you use the uuu built-in emmc_all option to burn the image?

Sincerely,

Han XU

landonh12, 2024-05-08:
Hi @allenxh,

Yes, emmc_all was used. Here is the actual command that was run:
```sudo uuu -b emmc_all imx-boot-imx8ulp-lpddr4-evk-sd.bin-flash_singleboot_m33 vinoth/imx-image-multimedia-imx8ulp-lpddr4-evk.rootfs-20240501134955.wic.zst```

allenxh, 2025-06-07:
> PS D:\LF_v6.1.1_1.0.0_images_IMX93EVK> uuu -lsusb
> uuu (Universal Update Utility) for nxp imx chips --
> libuuu_1.5.182-0-gda3cd53
>
> Connected Known USB Devices
> Path Chip Pro Vid Pid BcdVersion Serial_no
> ====================================================================
> 1:5 MX93 SDPS: 0x1FC9 0x014E 0x0001 3EA207D5E6C84739
>
> PS D:\LF_v6.1.1_1.0.0_images_IMX93EVK> uuu -b emmc
> imx-image-multimedia-imx93evk.wic
> uuu (Universal Update Utility) for nxp imx chips --
> libuuu_1.5.182-0-gda3cd53
>
> Success 0 Failure 1
>
> 1:5-3EA207D5 1/ 1 [HID(W): LIBUSB_ERROR_IO (-1) ] SDPS: boot -f
> imx-image-multimedia-imx93evk.wic
>
> i can't able to flash on emmc - MCIMX93EVK

-b emmc_all

Sincerely,

Han XU

nxpfrankli, 2025-06-10:
look like your imx93 chip's lifecyle fuse in your boards or ddr type is not matched bootloader in  imx-image-multimedia-imx93evk.wic