nxp-imx/mfgtools #425: uuu issue not able to flash u-boot image

pullannagari-babu, 2024-05-17:
I'm trying to upgrade the u-boot to 2024.04 from 2020.04 im 

C:\Users\FUP3KOR\Documents\imx8mm_evk\spl-added\RELEASE_FLASHING_PROD>uuu.exe -V imx8mm_flash_signed.uuu                                                                                               uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.4.43-0-ga9c099a                                                                                                                                                                                                                                                                                                                                                          Build in config:                                                                                                                                                                                                           Pctl     Chip            Vid     Pid     BcdVersion                                                                                                                                                                ==================================================                                                                                                                                                                 SDPS:    MX8QXP          0x1fc9  0x012f  [0x0002..0xffff]                                                                                                                                                          SDPS:    MX8QM           0x1fc9  0x0129  [0x0002..0xffff]                                                                                                                                                          SDPS:    MX8DXL          0x1fc9  0x0147                                                                                                                                                                            SDPS:    MX28            0x15a2  0x004f                                                                                                                                                                            SDPS:    MX815           0x1fc9  0x013e                                                                                                                                                                            SDPS:    MX865           0x1fc9  0x0146                                                                                                                                                                            SDP:     MX7D            0x15a2  0x0076                                                                                                                                                                            SDP:     MX6Q            0x15a2  0x0054                                                                                                                                                                            SDP:     MX6D            0x15a2  0x0061                                                                                                                                                                            SDP:     MX6SL           0x15a2  0x0063                                                                                                                                                                            SDP:     MX6SX           0x15a2  0x0071                                                                                                                                                                            SDP:     MX6UL           0x15a2  0x007d                                                                                                                                                                            SDP:     MX6ULL          0x15a2  0x0080                                                                                                                                                                            SDP:     MX6SLL          0x1fc9  0x0128                                                                                                                                                                            SDP:     MX7ULP          0x1fc9  0x0126                                                                                                                                                                            SDP:     MXRT106X        0x1fc9  0x0135                                                                                                                                                                            SDP:     MX8MM           0x1fc9  0x0134                                                                                                                                                                            SDP:     MX8MQ           0x1fc9  0x012b                                                                                                                                                                            SDPU:    SPL             0x0525  0xb4a4  [0x0000..0x04ff]                                                                                                                                                          SDPV:    SPL1            0x0525  0xb4a4  [0x0500..0x9998]                                                                                                                                                          SDPU:    SPL             0x0525  0xb4a4  [0x9999..0x9999]                                                                                                                                                          SDPU:    SPL             0x3016  0x1001  [0x0000..0x04ff]                                                                                                                                                          SDPV:    SPL1            0x3016  0x1001  [0x0500..0x9998]                                                                                                                                                          FBK:                     0x066f  0x9afe                                                                                                                                                                            FBK:                     0x066f  0x9bff                                                                                                                                                                            FB:                      0x0525  0xa4a5                                                                                                                                                                            FB:                      0x18d1  0x0d02                                                                                                                                                                            FB:                      0x3016  0x0001                                                                                                                                                                    Wait for Known USB Device Appear...                                                                                                                                                                                New USB Device Attached at 2:14                                                                                                                                                                                    2:14>Start Cmd:SDP: boot -f flash_signed.bin                                                                                                                                                                       6400%libusb: error [windows_transfer_callback] detected I/O error 1167: [1167] The device is not connected.                                                                                                        libusb: error [hid_submit_bulk_transfer] HID transfer failed: [1167] The device is not connected.                                                                                                                  2:14>Okay (0.547s)                                                                                                                                                                                                 libusb: wa [... cut ...]
FB: ucmd if test "${emmc_skip_fb}" != "yes"; then run emmc_cmd; fi
FB: ucmd setenv emmc_cmd mmc bootbus ${emmc_dev} 2 2 1;
FB: ucmd if test "${emmc_skip_fb}" != "yes"; then run emmc_cmd; fi

FB: ucmd setenv fastboot_buffer ${loadaddr}
FB: download -f Image_signed.bin
FB: ucmd setenv fastboot_buffer ${fdt_addr}
FB: download -f imx8mm-evk.dtb
FB: ucmd setenv fastboot_buffer ${initrd_addr}
FB: download -f fsl-image-mfgtool-initramfs-imx8mm-evk.cpio.gz.u-boot
FB: acmd ${kboot} ${loadaddr} ${initrd_addr} ${fdt_addr}

nxpfrankli, 2024-05-17:
New uboot and SPL change PID and VID because old one wrong use mass storage PID.

Please update uuu to latest one

pullannagari-babu, 2024-05-22:
Thank you so much @nxpfrankli its working

closed 2024-05-22 by pullannagari-babu