nxp-imx/mfgtools #429: Getting error with uuu on a Macbook Pro M2 machine: image file not found

michelgokan, 2024-06-25:
I'm trying to run the following command on my M2 machine:

```
sudo uuu -v -V ./full_image.uuu
```

Getting following error:

```
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.5.165-0-g7347a80

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX8DXL		 0x1fc9	 0x0147
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDPS:	 MX865		 0x1fc9	 0x0146
	SDPS:	 MX8ULP		 0x1fc9	 0x014a
	SDPS:	 MX8ULP		 0x1fc9	 0x014b
	SDPS:	 MX93		 0x1fc9	 0x014e
	SDPS:	 MX95		 0x1fc9	 0x015d
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPV:	 SPL1		 0x1fc9	 0x0151	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	SDPU:	 SPL		 0x3016	 0x1001	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x3016	 0x1001	 [0x0500..0x9998]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FBK:	 		 0x1fc9	 0x0153
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
	FB:	 		 0x3016	 0x0001
	FB:	 		 0x1fc9	 0x0152
	FB:	 		 0x0483	 0x0afb
Wait for Known USB Device Appear...

Error: FB: image file not found
```

Here is the content of the folder I'm in:

```
~/Downloads/846/mfgtool-files-portenta-x8 ls -lu
total 71960
-rw-r--r--@ 1 michelgokan  staff       587 25 Jun 17:33 bootloader.uuu
-rw-r--r--@ 1 michelgokan  staff  26945618 25 Jun 17:33 fitImage-portenta-x8-mfgtool
-rw-r--r--@ 1 michelgokan  staff       658 25 Jun 18:43 full_image.uuu
-rw-r--r--@ 1 michelgokan  staff    215591 25 Jun 17:33 imx-boot-mfgtool
-rw-r--r--@ 1 michelgokan  staff       511 25 Jun 17:33 linux_initramfs.uuu
-rw-r--r--@ 1 michelgokan  staff       180 25 Jun 17:33 probe_emmc.uuu
-rw-r--r--@ 1 michelgokan  staff       180 25 Jun 17:33 probe_sdcard.uuu
-rw-r--r--@ 1 michelgokan  staff      1043 25 Jun 17:33 test_ram.uuu
-rw-r--r--@ 1 michelgokan  staff   1416746 25 Jun 17:33 u-boot-mfgtool.itb
-rwxr-xr-x@ 1 michelgokan  staff   5632296 25 Jun 17:33 uuu
-rw-r--r--@ 1 michelgokan  staff   1282560 25 Jun 17:33 uuu.exe
-rwxr-xr-x@ 1 michelgokan  staff    664568 25 Jun 18:36 uuu_mac
```

I'm running with my local `uuu` installation that I installed using `brew install uuu`. Running the provided `uuu_mac` file is not working and I'm getting other type of errors:

```
./uuu_mac full_image.uuu
dyld[13729]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <244D7715-E5DC-3962-A45D-A562D5DED2A6> /Users/michelgokan/Downloads/846/mfgtool-files-portenta-x8/uuu_mac
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
[1]    13729 abort      ./uuu_mac full_image.uuu
```

Any help would be appreciated.

P.S: For more context, I'm trying to flash my Portenta X8 by following this tutorial: https://docs.arduino.cc/tutorials/portenta-x8/image-flashing/

nxpfrankli, 2024-06-25:
Did you install libusb?

michelgokan, 2024-07-29:
As a temporary solution, I installed Windows 11 using Parallels Desktop and managed to install the update via a VM.