nxp-imx/mfgtools #433: UBIFS image support?

swechencheng, 2024-08-12:
Hi,
I am wondering if there is support to directly transfer the whole ubifs image through `uuu`.
I build the rootfs by using `ubifs` for `SOC_DEFAULT_IMAGE_FSTYPES` in Yocto.
The product is a custom board based on iMX6UL with raw NAND Spansion S34ML04G200BHI (SLC).
I tried something like:
```
# burn rootfs
FBK: ucmd source /tmp/mtd.sh; flash_erase /dev/mtd${nandrootfs} 0 0
FBK: ucmd mkfifo /tmp/ubifs_pipe
FBK: acmd source /tmp/mtd.sh; ubiformat /dev/mtd${nandrootfs} -f /tmp/ubifs_pipe
FBK: ucp _rootfs t:/tmp/ubifs_pipe
```
And no surprise it does not work.
So, is there any way to transfer large image into FBK?
Thanks in advance.
BR/Chencheng

allenxh, 2024-08-12:
The file size is too large. I am not sure if you must generate the
single large ubifs image or if you can use the rootfs tarball with
following commands.

# burn rootfs
FBK: ucmd source /tmp/mtd.sh; flash_erase /dev/mtd${nandrootfs} 0 0
FBK: ucmd source /tmp/mtd.sh; ubiattach /dev/ubi_ctrl -m ${nandrootfs}
FBK: ucmd source /tmp/mtd.sh; ubimkvol /dev/ubi0 -Nnandrootfs -m
FBK: ucmd source /tmp/mtd.sh; mkdir -p /mnt/mtd
FBK: ucmd source /tmp/mtd.sh; mount -t ubifs ubi0:nandrootfs /mnt/mtd
FBK: acmd export EXTRACT_UNSAFE_SYMLINKS=1; tar -jx -C /mnt/mtd
FBK: ucp _rootfs.tar.bz2 t:-
FBK: sync
FBK: ucmd umount /mnt/mtd

swechencheng, 2024-08-12:
> The file size is too large. I am not sure if you must generate the single large ubifs image or if you can use the rootfs tarball with following commands. # burn rootfs FBK: ucmd source /tmp/mtd.sh; flash_erase /dev/mtd${nandrootfs} 0 0 FBK: ucmd source /tmp/mtd.sh; ubiattach /dev/ubi_ctrl -m ${nandrootfs} FBK: ucmd source /tmp/mtd.sh; ubimkvol /dev/ubi0 -Nnandrootfs -m FBK: ucmd source /tmp/mtd.sh; mkdir -p /mnt/mtd FBK: ucmd source /tmp/mtd.sh; mount -t ubifs ubi0:nandrootfs /mnt/mtd FBK: acmd export EXTRACT_UNSAFE_SYMLINKS=1; tar -jx -C /mnt/mtd FBK: ucp _rootfs.tar.bz2 t:- FBK: sync FBK: ucmd umount /mnt/mtd
> […](#)
> -- Sincerely, Han XU

Hi thanks for your reply.
I know tar ball works and I have succeeded on that. And I understand the tar ball gives us flexibility on strict raw NAND partition.
I am seeking the possibility of new way of packaging whole image, so that does not break our existing production line too much.
BR/Chencheng

swechencheng, 2024-08-13:
Thanks Allen,
But no luck:
```
2:411-0000000000000000>Start Cmd:FBK: acmd source /tmp/mtd.sh; ubiformat /dev/mtd${nandrootfs} -f -
2:411-0000000000000000>Okay (0.052s)
2:411-0000000000000000>Start Cmd:FBK: ucp _rootfs t:-
ubiformat: mtd5 (nand), size 513802240 bytes (490.0 MiB), 3920 eraseblocks of 131072 bytes (128.0 KiB), min. I/O size 2048 bytes
0%bscan: scanning eraseblock 1347 -- 34 % complete
```
It gets stuck at 0%.

allenxh, 2024-08-15:
If you boot to the kernel and burn this ubifs image(or a smaller one), does
it work?

swechencheng, 2024-08-15:
Currently not possible. If I boot into a rootfs, I am not able to unmount the ubi partition to run nand format.
Or do you mean run another initramfs via serial download with nand format in its init script? I doubt if the size of initramfs will be too big to be transferred through serial...

swechencheng, 2025-02-17:
We have successfully trimmed our rootfs so that the we can compile a ubi image properly with mkfs.ubifs and ubinize command with correct arguments. The image is under 70MB.
And in uuu script we can do:
```
FBK: ucp _rootfs t:/tmp/rfsimg
FBK: ucmd source /tmp/mtd.sh; flash_erase /dev/mtd${nandrootfs} 0 0
FBK: ucmd source /tmp/mtd.sh; ubiformat /dev/mtd${nandrootfs} -f /tmp/rfsimg
FBK: sync
```

closed 2025-02-17 by swechencheng